RIOT-OS/RIOT #4136: Support for MSP430X

l3nko, 2015-10-22:
Hi, I'm trying to use msp430 with support for 20-bit registers and a 20-bit address space to use all rom on platforms like z1. I was inspired by [contiki MSP430X](https://github.com/contiki-os/contiki/wiki/MSP430X) .
I use msp430-gcc v4.7.2 and add this flags:
>-mmemory-model=large -mcode-region=far -mdata-region=far -msr20 -mc2

During the compilation I fix some errors, for example at line 64 in cpu/msp430-common/cpu.c
```
unsigned short *stackptr = (unsigned short*) stk; 
```
compile get an error for casting pointer from integer of differenet size, so i add the cast to uintptr_t
```
unsigned short *stackptr = (unsigned short*) (uintptr_t) stk; 
```

Then I've used [z1-bsl-nopic](https://github.com/adamdunkels/contiki-fork/blob/master/tools/z1/z1-bsl-nopic) to program a z1, but now RIOT go in a reboot loop.
I debug and the problem seems to be in last function called from kernel_init because riot create the threads main and idle but after it reboot.

malosek, 2016-02-25:
Hello,

have had look at the recent msp430-common/cpu.c and for me seems like it can not work on 20bits msp-s at the mo.
or is it supposed to work?

wbr
malo

OlegHahm, 2016-02-25:
No, 20 bit MSPs are not supported right now. Feel free to add support. :)

malosek, 2016-03-02:
Hello,

Started to work on this, seems doable for the moment. Is anybody else playing with this subject, that we can join the forces?

wbr
malo

miri64, 2016-10-18:
Any progress?

malosek, 2016-10-20:
Hello,

have it done for TI compiler, but had to make additional changes since TI msp430 compiler does not support "naked" functions which is quite important compiler feature for porting riot...
Seems working well, device is up and running since then.
Anyway if somebody is interested I can provide CCS workspace with working project.
Do not have plans/time to port it to free compiler.

wbr
malo

squeekeek, 2017-06-23:
Hi everyone! I am also interested in adding MSP430X support in RIOT since we need a mote that has a large memory for our experiments. Is it possible to get a hold of the workspace so that I can continue porting it?

closed 2019-09-10 by stale[bot]