RIOT-OS/RIOT #4140: boards/pba-d-01-kw2x: ADC channel mode

PeterKietzmann, 2015-10-23:
According to the [periph_conf.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/pba-d-01-kw2x/include/periph_conf.h#L140) there is a single channel mode on the phytec board, which prevents to use `ADC_0_CH0` and `ADC_0_CH1` separately, when set. Running the periph_adc test shows, that a signal at `ADC_0_CH0_PIN` affects  both `ADC_0_CH0` and `ADC_0_CH1`. A signal at  `ADC_0_CH1_PIN` does nothing. So I assume we actually are in single mode. How can I change that mode? (I mean as a user, not as a developer)

miri64, 2016-10-18:
@jfischer-phytec-iot ping?

OlegHahm, 2017-01-15:
Who else has the hardware to test it?

PeterKietzmann, 2017-01-15:
At least the config file has changed and the above mentioned hint disappeared from that file. I can check this again tomorrow.

PeterKietzmann, 2017-01-16:
This is still an open question. `ADC_0_CH0` and`ADC_1_CH1` now refer to 
```
    { ADC0, GPIO_PIN(PORT_E, 2), 1 },
    { ADC0, GPIO_PIN(PORT_E, 3), 1 },
```
I would vote to either remove the ADC channel on `PTE3` (who ever used 6 ADC channels simultaneously?) or provide a switch plus explanation about this mode.

aabadie, 2018-04-17:
@PeterKietzmann is there still an issue here (since kinetis was refactored a lot recently) ? Maybe @gebart could help ?

gebart, 2018-04-17:
It's still an issue, but we need to examine the pba-d-01-kw2x schematics to see what the best configuration could be here.

kYc0o, 2018-04-17:
> we need to examine the pba-d-01-kw2x schematics to see what the best configuration could be here.

Once we tried to get them (@cgundogan  remember?) to add the button driver. We didn't receive any answer so we did it by looking directly at the board's lines and making tests with the ohmeter... A bit hard but not impossible.