RIOT-OS/RIOT #4142: iotlab-m3: packet loss increased

OlegHahm, 2015-10-23:
In comparison to the release version of RIOT the packet loss on iotlab-m3 (tested in the IoT-LAB testbed) has increased drastically. While the packet loss for pinging with 0 payload on the default channel between two nodes used to be close to 0% in the release version, it is about 10% on current master. I haven't checked which changes have been merged or bisected, but I would assume that some of the at86rf2xx driver changes is to blame.

cgundogan, 2015-10-23:
Are you sure it isn't just an interference problem because of other experiments? At my last attempt before the maintenance down-time I had a packet loss of 50% while pinging. After the maintenance it was back to "normal" to a near-zero value. I can have a look at it again later and see if I can "reprocude" the ~10%

OlegHahm, 2015-10-23:
Yes, I'm pretty sure about this. Tested several times on the same nodes with current master and release version.

OlegHahm, 2015-10-23:
bisecting revealed that the problem seems to be introduced by 5aeeabf4a9eaab3f25e67aeca6e4645d7ac84da6

OlegHahm, 2015-10-23:
Reverting the commit also eliminates the problem.

daniel-k, 2015-10-23:
Makes sense. I introduced this because incoming packets could block you from sending. Otherwise you can nit guarantee any timing. If you care for those packets this clearly is a problem. I'm without a laptop for the weekend, so I can't help too much. Are there any means in the interface to check for ongoing receptions?

But on the other side, is this something that should be handled in the driver itself or maybe one layer above?

OlegHahm, 2015-10-23:
Maybe I didn't understand #3933 correctly. I thought the received packet would have been discarded anyway if there's someone calling a TX operation at reception time. But these results seem to indicate that the packet is properly received. In this case, I think we should revert the change.

Concerning the question, the preloading option was added exactly for the purpose of time critical MAC protocols. IMO the MAC should use preloading and `NETOPT_STATE` with `NETOPT_STATE_TX` to send if timing is critical. Only in this case the RX operation should be canceled.

daniel-k, 2015-10-25:
@OlegHahm 
How is pinging implemented in this case? Is it only wait for a low-level ACK? Apart from this PR #3729 also got merged. When you cancel an ongoing RX transaction, the packet won't be ACKed so the sender will get a `NETDEV_EVENT_TX_NOACK` response instead of `NETDEV_EVENT_TX_COMPLETE`. Maybe if this is not handled, this could lead to the increased packet loss. You could try to revert #3729 instead and see if the behaviour stays the same.

OlegHahm, 2015-10-26:
> How is pinging implemented in this case? Is it only wait for a low-level ACK?

No, it is a real ICMP echo request/reply exchange on layer 3.

> Apart from this PR #3729 also got merged. 

But bisecting and reverting revealed that https://github.com/RIOT-OS/RIOT/commit/5aeeabf4a9eaab3f25e67aeca6e4645d7ac84da6 is the game changer here.

daniel-k, 2015-10-26:
I don't really understand how this can lead your observed behaviour (alone). Can you point me to the code you used for testing?

gebart, 2015-10-26:
@daniel-k I have a hypothesis what is happening...

In the original (pre 5aeeabf) implementation:
a packet is being received when at86rf2xx_tx_prepare is called, during the spin on line 218 the RX finishes => the RX interrupt is asserted => the ISR runs and passes a message to the at86rf2xx driver thread to read the frame => context switch from the thread which started at86rf2xx_tx_prepare to the driver thread => RX is either handled immediately or queued properly somewhere on the MCU side (at least no longer in radio on-chip frame buffer) => context switch back to the thread which was running at86rf2xx_tx_prepare => TX.

in the new implementation:
A packet is being received when the at86rf2xx_tx_prepare function is called => TRX_OFF => RX packet discarded => TX continues.

daniel-k, 2015-10-27:
> a packet is being received when at86rf2xx_tx_prepare is called, during the spin on line 218 the RX finishes => the RX interrupt is asserted => the ISR runs and passes a message to the at86rf2xx [...]

`at86rf2xx_tx_prepare()` should not be called outside of the driver, so effectively it's running in MAC context. On the other hand the message sent by the driver ISR is also sent so the MAC thread, so there should be no preemption/context switch.

@OlegHahm ping
Which code did you use to carry out your tests?

OlegHahm, 2015-10-27:
`gnrc_networking` example

OlegHahm, 2015-11-11:
@daniel-k, could you verify?

daniel-k, 2015-11-11:
@OlegHahm 
Not yet, sorry :( I'll probably find some time later today, but can't promise. Time is too scarce at the moment.

daniel-k, 2015-11-11:
Ran a test between two `samr21-xpro` lying next to each other on a table with recent `master` and `gnrc_networking` example. 1000 pings, 0 payload, 50ms interval => 0% packet loss:

```
ping6 1000 fe80::5855:6864:8d10:4576 0 50
...
2015-11-11 20:40:31,650 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=84 seq=996 hop limit=hu time = 64.009 ms
2015-11-11 20:40:31,716 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=84 seq=997 hop limit=hu time = 64.008 ms
2015-11-11 20:40:31,781 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=84 seq=998 hop limit=hu time = 64.007 ms
2015-11-11 20:40:31,848 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=84 seq=999 hop limit=hu time = 64.008 ms
2015-11-11 20:40:31,913 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=84 seq=1000 hop limit=hu time = 64.007 ms
2015-11-11 20:40:31,918 - INFO # --- fe80::5855:6864:8d10:4576 ping statistics ---
2015-11-11 20:40:31,924 - INFO # 1000 packets transmitted, 1000 received, 0% packet loss, time 66.06120133 s
2015-11-11 20:40:31,928 - INFO # rtt min/avg/max = 6.406/8.659/10.874 ms
```

Not related to this topic, but the ping time seems wrong. It is around 64ms all the time, but `rtt min/avg/max = 6.406/8.659/10.874 ms` seems correct as wireshark has similar numbers.

OlegHahm, 2015-11-11:
> Not related to this topic, but the ping time seems wrong. It is around 64ms all the time, but rtt min/avg/max = 6.406/8.659/10.874 ms seems correct as wireshark has similar numbers.

Might be related to your toolchain. Prrinting of short unsigned integers seems to be non-working neither. See #4051.

OlegHahm, 2015-11-11:
Try without delay between the pings.

daniel-k, 2015-11-11:
```
ping6 1000 fe80::5855:6864:8d10:4576 0 0
...
2015-11-11 21:22:11,147 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=83 seq=997 hop limit=hu time = 64.007 ms
2015-11-11 21:22:11,164 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=83 seq=998 hop limit=hu time = 64.009 ms
2015-11-11 21:22:11,181 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=83 seq=999 hop limit=hu time = 64.010 ms
2015-11-11 21:22:11,199 - INFO # 8 bytes from fe80::5855:6864:8d10:4576: id=83 seq=1000 hop limit=hu time = 64.010 ms
2015-11-11 21:22:11,204 - INFO # --- fe80::5855:6864:8d10:4576 ping statistics ---
2015-11-11 21:22:11,210 - INFO # 1000 packets transmitted, 999 received, 1% packet loss, time 17.06128832 s
2015-11-11 21:22:11,214 - INFO # rtt min/avg/max = 6.400/8.652/14.619 ms
```

OlegHahm, 2015-11-11:
Strange, I still have about 5% packet loss in the single hop scenario.

daniel-k, 2015-11-11:
Can you tell what happens when a ping timeouts? For me it happens when the sender has sent its request and then decides to send a router solicitation. The receiver replies 4 times, but the sender doesn't seem to get it => timeout. But that only happens for 1 in 1000 packets as you can see above.

cgundogan, 2015-11-11:
single-hop, iotlab-testbed, site=grenoble, gnrc_networking example:
```
ping6 1000 fe80::3432:4833:46d5:9d12 0 0
1447273961.668521;m3-7;--- fe80::3432:4833:46d5:9d12 ping statistics ---
1447273961.669462;m3-7;1000 packets transmitted, 927 received, 8% packet loss, time 94.06516315 s
1447273961.670336;m3-7;rtt min/avg/max = 4.598/19.752/46.167 ms
```

and again because it's fun:

```
ping6 1000 fe80::3432:4833:46d5:9d12 0 0
1447274090.679040;m3-7;--- fe80::3432:4833:46d5:9d12 ping statistics ---
1447274090.680747;m3-7;1000 packets transmitted, 921 received, 8% packet loss, time 100.06657308 s
1447274090.682559;m3-7;rtt min/avg/max = 4.288/19.900/50.186 ms
```

OlegHahm, 2015-11-11:
I don't have a sniffer running, but within this short amount of time there should be sent only a few RS, while I loose about 50 packets.

daniel-k, 2015-11-11:
@cgundogan 
Hmm, you have much higher avg and max values. Is there any other traffic on the same channel?

OlegHahm, 2015-11-11:
I have similar values for the round trip times.

OlegHahm, 2015-11-11:
While testing in a different site (rocquencourt).

cgundogan, 2015-11-11:
@daniel-k may be possible. I have no idea how crowded the grenoble site is currently. But ~8% packet loss is still interestingly high

daniel-k, 2015-11-11:
@cgundogan 
Can you please change the network ID and maybe also the channel just for testing?

OlegHahm, 2015-11-11:
```
1447274558.127119;m3-14;1000 packets transmitted, 999 received, 1% packet loss, time 9.06339489 s
1447274558.127232;m3-14;rtt min/avg/max = 4.268/6.500/8.678 ms
```
with 5aeeabf4a9eaab3f25e67aeca6e4645d7ac84da6 reverted.

```
1447274712.935015;m3-14;1000 packets transmitted, 959 received, 5% packet loss, time 56.06295354 s
1447274712.936164;m3-14;rtt min/avg/max = 4.290/13.504/50.142 ms
```
on master

OlegHahm, 2015-11-11:
Similar loss rate on channel 15.

cgundogan, 2015-11-11:
I switched to channel 15:
```
1447274819.449247;m3-7;--- fe80::3432:4833:46d5:9d12 ping statistics ---
1447274819.451277;m3-7;1000 packets transmitted, 913 received, 9% packet loss, time 108.06229891 s
1447274819.451552;m3-7;rtt min/avg/max = 4.597/19.482/52.027 ms
```

Set to channel 11:
```
1447274938.290541;m3-7;--- fe80::3432:4833:46d5:9d12 ping statistics ---
1447274938.292543;m3-7;1000 packets transmitted, 927 received, 8% packet loss, time 94.06466339 s
1447274938.293404;m3-7;rtt min/avg/max = 4.287/19.702/56.998 ms
```

OlegHahm, 2015-11-11:
> Hmm, you have much higher avg and max values. Is there any other traffic on the same channel?

I haven't checked, but maybe failing pings get in these stats, too. Or it is because of a higher retry count on the link layer.

cgundogan, 2015-11-11:
@daniel-k you test on samr21-xpros locally, right?

daniel-k, 2015-11-11:
> @daniel-k you test on samr21-xpros locally, right?

Yep.

daniel-k, 2015-11-11:
So I was staring at the diff of 5aeeabf4 for quite some time now and studied [this nice migration guide from at86rf231 to at86rf233](http://www.atmel.com/Images/Atmel-42221-Migration-from-AT86RF231-to-AT86RF233_AP-Note_AT02602.pdf) and I still can't think of a reason that would explain the observed behaviour.

OlegHahm, 2015-11-12:
Maybe we should try this issue the other way around: did anyone (@daniel-k) observed a performance increase from https://github.com/RIOT-OS/RIOT/commit/5aeeabf4a9eaab3f25e67aeca6e4645d7ac84da6?

OlegHahm, 2015-11-12:
But I can verify that the problem doesn't seem to occur on the SAM-R21.

thomaseichinger, 2015-11-13:
So I ran some additional local only tests which support the findings so far.
* rf231 with the abortion of RX has about 3-4% packet loss while it has 0-1% packets lost out of 10000 without.
* rf233 seems to be fine with and without RX abortion.
* the abortion only effects the pinged/receiving node
* what I observed from debugging using GPIO toggling, adding NOPs after `at86rf2xx_force_trx_off(dev)` and `at86rf2xx_set_state(dev, state)`:L237 results in also almost no packet loss.

I think the main reason why we see these different results comes from the changed state transition timing between the two versions. Transition from `TRX_OFF` to `PLL_ON` is cut by one third for the rf233 with 80us while rf231 needs 110us. But that's only speculation by now.

daniel-k, 2015-11-13:
@thomaseichinger 
Thanks for investigating! How many `nop`s did you add? I don't understand yet why this changes anything. Both functions busy-wait for the state transisition so it shouldn't make a difference.
Maybe there is some compiler optimization going on, but I'm just guessing. 

Did you test on iotlab, too? If not, which platform? Did you capture any wireless traffic by chance?

OlegHahm, 2015-11-13:
Another interesting observation: current master has a packet loss of up to 50% and a RTT of about 90ms. Commenting out the `at86rf2xx_force_trx_off(dev)` line decreases RTT to about 25ms and eliminates the packet loss.

thomaseichinger, 2015-11-13:
@OlegHahm I think you referencing rf212b with your last comment, right?

thomaseichinger, 2015-11-13:
@daniel-k My tests are mostly concerning rf231 with iotlab-m3 nodes. I didn't find the right `nop`s number yet, I can see it with 3 consecutive calls to `gpio_set` after forcing off and `gpio_clear` after setting the TX state.

daniel-k, 2015-11-13:
Sorry guys, but without a packet dump this is only guesswork :/ My iotlab account still isn't approved yet.

OlegHahm, 2015-11-13:
Maybe I can summon @cladmi to help speeding up the process. ;-)

authmillenon, 2015-11-13:
Accio @cladmi! (sorry, could not resist ^^)

thomaseichinger, 2015-11-13:
@daniel-k [Here you go.](https://www.cloudshark.org/captures/6db8d11af5c5) Wireshark dump from a samr21-xpro to a iotlab-m3.

daniel-k, 2015-11-13:
> @daniel-k [Here you go.](https://www.cloudshark.org/captures/6db8d11af5c5) Wireshark dump from a samr21-xpro to a iotlab-m3.

Thanks a lot. First thing that seems strange: why does `:c9b6` send requests so quick after each other? Sometimes it's less than 1.5ms between two requests. Same goes for the reply. That seems clearly wrong to me. Maybe lost ACKs?

Second, there are only some really rare ACKs in the packet dump. Where are all the other ACKs that should be there? 

Some questions to this setup:
* `:c9b6` -> `samr21-xpro` and `:8172` -> `iotlab-m3`, right?
* Both nodes on current `master`?
* Which interval between pings did you choose?
* How is the packet dump generated? Can I trust the timestamps? How big is the inaccuracy?

Sorry for bothering you that much. It would be much easier if I had acces to those transceivers myself :/

daniel-k, 2015-11-13:
Just to state more observation. Let's look at packets 562 - 565.

562: request seq=190
563: reply seq=190 1.5ms after 562
564: 2nd reply seq=190 16ms after 562
565: next request seq=191 1s after 562, so clearly ping timeout -> 563+564 not received.

At least 564 should have been received. As the pinger didn't resend the request I assume it was properly ACKed by the pingee. I will investigate this further, just wanted to share and document progress :)

OlegHahm, 2015-11-14:
@thomaseichinger, did you use a RIOT node for sniffing?

cladmi, 2015-11-16:
Account validated.

Regarding radio sniffer, we have the issue on IoT-LAB that ACKs packets don't get captured: https://github.com/iot-lab/iot-lab/issues/191
If you have something working for this I would love to see it.

OlegHahm, 2015-11-16:
@cladmi, thanks.  Actually, the last time I checked we had a similar problem with the RIOT based sniffer.

OlegHahm, 2015-11-22:
Should be fixed with https://github.com/RIOT-OS/RIOT/issues/4262 being merged.

closed 2015-11-22 by OlegHahm