RIOT-OS/RIOT #4151: RFC: create lightweight tag "latest" pointing to the latest commit

authmillenon, 2015-10-26:
Just an idea I had, when I was updating the documentation. That would make creating links to the repository way easier.

OlegHahm, 2015-10-26:
I don't unerstand the idea.

Kijewski, 2015-10-26:
It's called [`master`](https://github.com/RIOT-OS/RIOT/tree/master)?!

kaspar030, 2015-10-26:
I thought the same...

authmillenon, 2015-10-26:
Sorry. Somehow I missed the most essential part of the sentence. I meant the latest commit on the most current release (branch).

kaspar030, 2015-10-26:
Wouldn't that require pushing the new tag after every commit?

Kijewski, 2015-10-26:
Why do you want to use a tag? Tags should be immutable and you have to force-pull them if they were changed AFAIK. The release branch has a name, too: [`2015.09-branch`](https://github.com/RIOT-OS/RIOT/tree/2015.09-branch).

cgundogan, 2015-10-26:
@Kijewski I assume @authmillenon's proposed `tag` points to the **latest** commit in the **latest** release. Once the release branch [https://github.com/RIOT-OS/RIOT/tree/2015.09-branch](https://github.com/RIOT-OS/RIOT/tree/2015.09-branch) is gone, then all links to that url will point into the void.

However, as @kaspar030 did point out: This "moving tag" requires manual interaction. Or could some kind of remote hook do this?

Kijewski, 2015-10-26:
@cgundogan, but why should we delete the branch? We can just stop updating it once we begin with the next release branch.

authmillenon, 2015-10-26:
> Wouldn't that require pushing the new tag after every commit?

Only if (as @cgundogan suggested offline, that's why I put it in here) you point it to the last commit, yes. If you just point it to the latest release you only have to push with every release.

> Why do you want to use a tag? Tags should be immutable and you have to force-pull them if they were changed AFAIK.

Not lightweight tags, to my knowledge. They are just references, not objects

>  The release branch has a name, too: 2015.09-branch.

But that name will change with the next release(s), forcing us to rewrite the doc with every release.

A `latest` tag would only require us to change that tag with every release, since we can refer everything in the documentation just with `latest`. E. g. (from the new main page currently in work)

```md
The quickest start {#the-quickest-start}
==================
You can run RIOT on most IoT devices, on open-access testbed hardware (e.g. IoT-lab), and also directly as a process on your Linux/FreeBSD/OSX machine (we call this the `native` port).
Try it right now in your terminal window:

~~~~~~~{.sh}
git clone git://github.com/RIOT-OS/RIOT.git # assumption: git is pre-installed
git checkout latest
./cpu/tapsetup/tapsetup                       # create virtual ethernet interfaces to connect to RIOT
cd examples/default/
make all
make term
\~~~~~~~
```

authmillenon, 2015-10-26:
> @cgundogan, but why should we delete the branch? We can just stop updating it once we begin with the next release branch.

It's not about removing the branch, the doc should just keep up with the current releases.

authmillenon, 2015-10-26:
> However, as @kaspar030 did point out: This "moving tag" requires manual interaction. Or could some kind of remote hook do this?

Might be possible.

Kijewski, 2015-10-26:
Well, where do you even need absolute names? You can use relative paths in Doxygen, can't you?

authmillenon, 2015-10-26:
Really? How would you do this in the example I gave?

authmillenon, 2015-10-26:
And how would you link to an example?

```md
Read the [README of the default example](https://github.com/RIOT-OS/RIOT/tree/latest/examples/default) for further information.
```

Is how I would do it then.

Kijewski, 2015-10-26:
Adding `README.md` to `doxygen/riot.doxyfile` and then linking with `[guckst du hier](./examples/default/README.md)` should work. Using absolute URLs when the content **will** change defeats the purse of tags.

gebart, 2015-10-27:
I agree with @Kijewski, tags should not be used this way, that's what branches are for.
if you want you could just create a branch `stable` (or `latest`) and let it follow the latest release branch (`2015.09-branch`), though you'll have to update it whenever PRs are merged to the release branch.

authmillenon, 2015-10-27:
So git-internally doing the exact same thing that you would do with a lightweight tag :-). But yeah, I guess your right semantically tags should stay immutable.

authmillenon, 2015-10-27:
> Adding README.md to doxygen/riot.doxyfile's INPUT and then linking with [guckst du hier](./examples/default/README.md) should work (untested). Using absolute URLs when the content will change defeats the purpose of tags.

That won't work since doxygen (intentionally) doesn't pick up every file and copies it into the HTML path. Also, if we pick it up by doxygen it would be rendered, so we either need to reference it (the labeling syntax for is incompatible to the GitHub markdown syntax btw) or drop markdown support for doxygen altogether. Also, I prefer to closer connect the doc.riot-os.org, the READMEs in the GitHub repository, and the GitHub wiki.

OlegHahm, 2015-10-27:
I would strongly advocate to put as little URLs as possible into Doxygen or any READMEs. This documentation should be useful and readable offline, too.

miri64, 2016-10-18:
Can be closed I think.

closed 2016-10-18 by miri64