RIOT-OS/RIOT #4169: SAMR21: GPIO init dir: out, low on init

bapclenet, 2015-10-27:
On the SAMR21, why do we set the pin to a low level on init? [1]

    port->OUTCLR.reg = pin_mask;            /* set pin LOW on init */

I would like my pin to be high after init otherwise it seems like I toogle my pin.

[1][https://github.com/RIOT-OS/RIOT/blob/master/cpu/samd21/periph/gpio.c#L122](https://github.com/RIOT-OS/RIOT/blob/master/cpu/samd21/periph/gpio.c#L122)

bapclenet, 2015-10-27:
At least we should let the choice to the user, if he wants:
 - PULLDOWN, low on init
 - PULLUP, high on init
 - NOPULL, low on init

I know why I have this pike, DIR:OUT, PULLUP, look (l103):

    switch (pushpull) {
            case GPIO_PULLDOWN:
                        port->OUTCLR.reg = pin_mask;
                        port->PINCFG[pin_pos].reg = PORT_PINCFG_PULLEN;
                        break;
            case GPIO_PULLUP:
                        port->OUTSET.reg = pin_mask;

We set it to high

                        port->PINCFG[pin_pos].reg = PORT_PINCFG_PULLEN;
                        break;
            case GPIO_NOPULL:
                        port->PINCFG[pin_pos].reg = 0;
                        break;
                }
                /* set pin_pos direction */
                if (dir == GPIO_DIR_OUT) {
                    if (pushpull == GPIO_PULLDOWN) {
                        return -1;
                    }
                    port->DIRSET.reg = pin_mask;            /* configure as output */
                    port->OUTCLR.reg = pin_mask;            /* set pin LOW on init */

And then to low

                }
                else {
                    port->DIRCLR.reg = pin_mask;            /* configure as input */
                    port->PINCFG[pin_pos].reg |= PORT_PINCFG_INEN;
                }

bapclenet, 2015-10-27:
I suggest something like this:

    switch (pushpull) {
        case GPIO_PULLDOWN:
            port->OUTCLR.reg = pin_mask;
            port->PINCFG[pin_pos].reg = PORT_PINCFG_PULLEN;
            break;
        case GPIO_PULLUP:
            port->OUTSET.reg = pin_mask;
            port->PINCFG[pin_pos].reg = PORT_PINCFG_PULLEN;
            break;
        case GPIO_NOPULL:
            port->OUTCLR.reg = pin_mask;
            port->PINCFG[pin_pos].reg = 0;
            break;
    }
    /* set pin_pos direction */
    if (dir == GPIO_DIR_OUT) {
        if (pushpull == GPIO_PULLDOWN) {
            return -1;
        }
        port->DIRSET.reg = pin_mask;            /* configure as output */
    }
    else {
        port->DIRCLR.reg = pin_mask;            /* configure as input */
        port->PINCFG[pin_pos].reg |= PORT_PINCFG_INEN;
    }

haukepetersen, 2015-10-27:
> On the SAMR21, why do we set the pin to a low level on init?

We do this on every board -> the reason is simple: after initialization, the pin is low, independent on the CPU you are working on.

bapclenet, 2015-10-27:
Ok ok, it just takes a bit more time to set it to 1 after initialization

bapclenet, 2015-10-27:
I close the issue as it's not an issue regarding RIOT gpio initialization

closed 2015-10-27 by bapclenet