RIOT-OS/RIOT #4204: tests/xtimer_*: add README

PeterKietzmann, 2015-11-02:
It would be nice to have a small README for all xtimer tests. Something like the idea about the test and a rough overview of the happening as well as a short description about outcome.

PeterKietzmann, 2016-03-18:
@A-Paul AFAIK you spend a lot of time on testing timers the last weeks. You think it's useful to have a small README or it it documented by now? Didn't check it but I think you know best what to say about these tests

miri64, 2016-10-18:
Ping?

OlegHahm, 2017-01-15:
@A-Paul, @PeterKietzmann, any progress?

PeterKietzmann, 2017-01-15:
Well, it was more meant as a feature **request** ;-)

PeterKietzmann, 2018-12-18:
This issue is rather old and was opened in the initial days of xtimer where documentation wasn't barely in place. Nowadays nearly all tests have a README. Tests that don't provide one have clear statements about success/failure by their testrunner -> I'm closing this issue now

closed 2018-12-18 by PeterKietzmann