RIOT-OS/RIOT #4206: zep example doesn't build

brummer-simon, 2015-11-02:
Hi Everybody, 

The HowTo for Zep, reachable under (https://github.com/RIOT-OS/RIOT/wiki/Testing-6LoWPAN-on-Ethernet-based-devices), doesn't compile. The Modules in the HowTo are still prefixed with ng instead of gnrc, if I try to build with USEMODULE += gnrc_zep, the function _zep_init() has an undefined reference to function gnrc_nomac_init().

authmillenon, 2015-11-02:
The Wiki should be editable by you. So if you notice anything deprecated, please don't hesitate to update it.

authmillenon, 2015-11-02:
About the nomac stuff: I will look into it.

authmillenon, 2015-11-02:
Ah, since `netdev2_gnrc` does not use nomac anymore you need to add it too to `USEMODULE`. But for some reason `zep_init` now hangs on initialization :-/.

authmillenon, 2015-11-02:
Okay… because both `gnrc_sixlowpan_nd` and `gnrc_ndp_host` are included https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L534 is always true and always the ZEP interface (because link-local addresses are always back-translated). This leads to an endless loop of ever growing encapsulation (ZEP-Interface sends RA, puts it into ZEP, sends ZEP packet over IP, IP forwards it to ZEP, ZEP puts it into ZEP, sends ZEP packet over IP, etc.). @OlegHahm it was your idea to exit early in this function. Any idea how to fix that (reversing them will also lead to sending it out over the ZEP, because there is nothing in the neighbor cache yet).

authmillenon, 2015-11-02:
@brummer-simon this problem is a little bit more complicated. Either you deactivate `gnrc_sixlowpan_nd` (which should work) or you have to test on real hardware.

miri64, 2016-10-18:
I think ZEP is another thing I have to look at ;-)

OlegHahm, 2017-01-15:
@miri64, I suppose you're not planning to fix this any more with the upcoming change of ZEP, right?

miri64, 2017-01-15:
Yep, the plan is to remove `gnrc_zep` as soon #6121 is merged

closed 2017-01-15 by OlegHahm