RIOT-OS/RIOT #4211: timex: macro names are missleading

authmillenon, 2015-11-02:
Am I the only one that thinks the name of `SEC_IN_USEC`, `SEC_IN_MS`, `MS_IN_USEC`, and `USEC_IN_NS` are somewhat missleading?

1. They are inconsistent in itself. Some abbreviate "seconds" with `SEC`, some with `S`
2. Their values reflect the opposite of their name: there are 1000000 microseconds in a second, not 1000000 seconds in a microsecond; there are 1000 milliseconds in a second, not 1000 seconds in a microsecond; etc.

BytesGalore, 2015-11-02:
I think it means **unit** representation in **other unit**, 
e.g. second represented in microseconds => `SEC_IN_USEC`

authmillenon, 2015-11-02:
If you have to make something up (that's what I'm interpreting from your "I think it means …" ;-)) it isn't intuitive ;-).

BytesGalore, 2015-11-02:
It felt correct for me but I was uncertain if I found the right terms to describe it :). 
But I agree with `S` vs `SEC` either the one or the other.
I vote to use the SI unit representation of time [1], so just `S`.

[1] http://physics.nist.gov/cuu/Units/units.html

OlegHahm, 2015-11-03:
:+1: for renaming `USEC` to `US`, everything else is intuitive for me and renaming `SEC` to `S` seems to be more ambiguous to me.

authmillenon, 2015-11-03:
Though the SI says otherwise I would prefer SEC, MSEC, USEC, NSEC, since US_IN_S looks a bit non-descriptive. @oleghahm can you explain the intuition? I still read it as SEC_IN_USEC = seconds in microseconds.

gebart, 2015-11-03:
@authmillenon if you write it in code as
```c
123 * SEC_IN_USEC
```
you can read it aloud as "123 seconds in microseconds"

OlegHahm, 2015-11-03:
exactly

authmillenon, 2015-11-03:
I find that less intuitive and less mathematically sound than "123 times microseconds in seconds"

Kijewski, 2015-11-03:
I'm with @authmillenon. In my head SEC_IN_USEC evaluates to 10^-6. There is 0.000001 seconds in a microsecond. But there are 10^6 microseconds in a second, so USEC_IN_SEC = 1000000.

I'd say: `#define CONV_S_US(X) (1000000ull * (X))`, convert seconds to microseconds.

OlegHahm, 2015-11-03:
Can you find something less speaking? ;)

OlegHahm, 2015-11-03:
Seems to be a personal thing: in my thinking a second is no container to be filled with mili- or microseconds.

Kijewski, 2015-11-03:
@OlegHahm, `#define WOULD_YOU_PLEASE_BE_SO_KIND_AS_TO_CONVERT_SECONDS_TO_MILLISECONDS_FOR_ME(…) …`?

OlegHahm, 2015-11-03:
awesome

kaspar030, 2015-11-03:
I'm with @Kijewski and @authmillenon here.

I think we've gotten used to the old defines, but it hard to argue that

```123 * USEC_PER_SEC``` would less intuitive than ```123 * SEC_IN_USEC```.

OlegHahm, 2015-11-03:
I'm fine with both as long as we don't go back to macro functions here.

kaspar030, 2015-11-03:
@Kijewski @authmillenon would you also prefer ```USEC_PER_SEC``` over ```USEC_IN_SEC```?

authmillenon, 2015-11-04:
I'm fine with either.

Kijewski, 2015-11-04:
I like `USEC_PER_SEC` :+1:

haukepetersen, 2016-04-20:
ongoing discussion, will hopefully be solved by 2016.07

kYc0o, 2016-07-26:
@gebart I have the impression that on one of your PRs you addressed this issue. Am I wrong?

gebart, 2016-07-26:
@kYc0o I don't remember changing this

kYc0o, 2016-07-26:
Ok let's postpone it then.

miri64, 2016-10-17:
Clearing milestone because there doesn't seem to be much traction behind this family of issues.

OlegHahm, 2017-01-15:
Was there a consensus?

PR #6399 (2017-01-19): timex: unambiguous time conversion macros
Fixes #4211.

I used:
```shell
git grep -l SEC_IN_USEC | xargs sed -i 's/MIN_IN_SEC/SEC_PER_MIN/'
git grep -l SEC_IN_USEC | xargs sed -i 's/SEC_IN_CS/CS_PER_SEC/'
git grep -l SEC_IN_USEC | xargs sed -i 's/SEC_IN_MS/MS_PER_SEC/'
git grep -l SEC_IN_USEC | xargs sed -i 's/SEC_IN_MS/MS_PER_SEC/'
git grep -l SEC_IN_USEC | xargs sed -i 's/MS_IN_USEC/USEC_PER_MS/'
git grep -l SEC_IN_USEC | xargs sed -i 's/USEC_IN_NS/NS_PER_USEC/'
```
closed 2017-01-19 by OlegHahm