RIOT-OS/RIOT #4214: ping6 missing from Routing/Networking Examples

msolters, 2015-11-02:
Due to the replacement of `vtimer` with `xtimer` in https://github.com/RIOT-OS/RIOT/commit/5fb104160edae3e0fe0d40c33268d3dff28d57a5, the `ping6` command is not being included in applications that expect it (such as the gnrc_border_router, gnrc_networking examples).  This is due to the module checking logic [here](https://github.com/RIOT-OS/RIOT/blob/7797bb438d6c421d2fd96afaa1011eb2195782be/sys/shell/commands/shell_commands.c#L93) and [here](https://github.com/RIOT-OS/RIOT/blob/7797bb438d6c421d2fd96afaa1011eb2195782be/sys/shell/commands/shell_commands.c#L185).

If I understand this correctly, we need to change those rules to rely on XTIMER, and therefore then to refactor [sc_icmpv6_echo.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/shell/commands/sc_icmpv6_echo.c) in terms of XTIMER rather than VTIMER.

authmillenon, 2015-11-03:
Should be fixed by #4210.

authmillenon, 2015-11-03:
In the meantime: just add vtimer to your application's modules.

msolters, 2015-11-03:
:100:

PeterKietzmann, 2015-12-04:
#4210 was merged so I'll close this one

closed 2015-12-04 by PeterKietzmann