RIOT-OS/RIOT #4220: Contiki devices communicate to other wireless network devices running RIOT OS

xianlin, 2015-11-03:
I posted this question in contiki github issue page as well.
https://github.com/contiki-os/contiki/issues/1345

Is it possible to let devices running RIOT communicate with Contiki devices?
There is a project `COAP server` ongoing to let these 2 OS talk to each other. 

I wonder if I put 2 devices with RIOT and Contiki together, if they didn't implement the network layers (6lowpan) in the same way, they cannot talk to each other.

For the communication to happen, the 6lowpan standard must be implemented the same way. Is it NOT happening now?

DipSwitch, 2015-11-03:
As far as I know, the RFC is followed. But some features are not yet implemented.

OlegHahm, 2015-11-04:
Not sure, I understand the question correctly. There are different layers of interoperability. 6LoWPAN and IPv6 between Contiki and RIOT should be interoperable. RPL in storing mode is currently missing (see #3662). For the 802.15.4 part, Contiki and RIOT should be interoperable when using the same RDC scheme.

gebart, 2015-11-04:
@OlegHahm RPL in _non-_ storing mode is the missing part. Storing mode is working, except for the issues outlined in #3662.

OlegHahm, 2015-11-04:
Eh, sorry, yes, I meant some things are missing for RPL storing mode.

gebart, 2015-11-04:
I have run a successful basic RPL interoperability test between Contiki `examples/ipv6/rpl_border_router` and RIOT `examples/gnrc_networking`

Some special configuration was necessary on the Contiki side to make RIOT accept the incoming traffic.

Add the following to Contiki `project-conf.h`:
```
#define RPL_CONF_INSERT_HBH_OPTION 0
#define RPL_CONF_OF rpl_of0
```

Start the border router connection on the PC:
```
tunslip6 -B 115200 -t contiki0 -L -v2 -s /dev/ttyUSBx fdfd::ff/64
```

Start a netcat udp receiver on the PC:
```
nc -6 -u -l -s fdfd::ff -p 12345 -v
```

The following commands were used in the RIOT shell: 
(`7` is the interface id found when running `ifconfig` without any arguments)
```
ifconfig 7 set chan 0
ifconfig 7 set page 2
ifconfig 7 set pan 0x777
rpl init 7
rpl
ping6 fe80::1
ping6 fdfd::1
ping6 fdfd::ff
fibroute
ifconfig
udp send fdfd::ff 12345 asdfg 1
```

`asdfg` shows up in the `nc` terminal.

OlegHahm, 2015-11-04:
Thanks for this information. We (@thomaseichinger and me) are gonna participate in an [ETSI plugtest](http://www.etsi.org/news-events/events/1003-6lo-plugtests) this weekend to test interoperability of various [6lo technologies](https://tools.ietf.org/wg/6lo/) between RIOT and other implementations. I don't know if some people from the Contiki community will be around.

OlegHahm, 2016-02-09:
Tested basic communication with Contiki over 6LoWPAN using RPL last week. Worked.

closed 2016-02-09 by OlegHahm

gebart, 2016-02-09:
@OlegHahm could you summarize the changes (if any) that you needed to make to Contiki in order to get communication going?
Like I wrote above, I had to change project-conf.h in Contiki in order for RIOT to accept Contiki's traffic, did you do the same changes?

OlegHahm, 2016-02-09:
I don't know, because I was only responsible for the RIOT site. @simonduq was responsible for the Contiki node. AFAIK the only thing that he changed was setting the RDC to NULLMAC (or however it is called). Maybe @kYc0o knows more details.

simonduq, 2016-02-09:
Setting the same MAC, PANID and channel. On the Riot side you changed something to include the PIO in every DIO, but that was only to make the network joining procedure faster. No other change I remember of. We had RPL storing mode working in a two nodes network, the routing table were correct at the root, and the nodes could ping one another.