RIOT-OS/RIOT #4222: POSIX sockets: implement (p)select

OlegHahm, 2015-11-04:
For the sake of ease porting libraries based on POSIX sockets (e.g. libcoap, ccn-lite...) an implementation of the `select()` function for synchronous I/O multiplexing would be very helpful.

Kijewski, 2015-11-04:
I would rather say [`poll()`](http://pubs.opengroup.org/onlinepubs/9699919799/functions/poll.html), because the API of [`select()`](http://pubs.opengroup.org/onlinepubs/9699919799/functions/select.html) is very difficult to use or even understand.

OlegHahm, 2015-11-04:
Nevertheless it is required for the above mentioned examples.

kaspar030, 2015-11-04:
#4103 might be used to efficiently implement this.

authmillenon, 2015-11-04:
Might it make sense to have that of conn, too?

OlegHahm, 2015-11-04:
@kaspar030, maybe.

@authmillenon, I sense that we need to rethink conn a little bit, but in general, yes.

kaspar030, 2017-10-24:
Closed in favour of #7801.

closed 2017-10-24 by kaspar030