RIOT-OS/RIOT #4231: shell: differing address formats

A-Paul, 2015-11-05:
In the shell `ifconfig` prints the IPv6 addresses with a trailing prefix number. The `ping6` command doesn't accept this and prints an "usage" messages with no specific hint what went wrong.
I think it would be less confusing if the reported format could also be used as input.

authmillenon, 2015-11-05:
Why? This would add additional code overhead and doesn't really make sense, since ping doesn't require the prefix nor does it know what to do with its length. What when the number provided by the user is wrong? Should this cause an error? Also: even Linux does not parse address + prefix (probably exactly for that reason):

```
$ ping6 fd53:5ee2:328:65:ea39:35ff:feec:f33a
PING fd53:5ee2:328:65:ea39:35ff:feec:f33a(fd53:5ee2:328:65:ea39:35ff:feec:f33a) 56 data bytes
64 bytes from fd53:5ee2:328:65:ea39:35ff:feec:f33a: icmp_seq=1 ttl=64 time=0.416 ms
64 bytes from fd53:5ee2:328:65:ea39:35ff:feec:f33a: icmp_seq=2 ttl=64 time=0.313 ms
^C
--- fd53:5ee2:328:65:ea39:35ff:feec:f33a ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 999ms
rtt min/avg/max/mdev = 0.313/0.364/0.416/0.054 ms
$ ping6 fd53:5ee2:328:65:ea39:35ff:feec:f33a/64
unknown host
```

A-Paul, 2015-11-05:
Sorry for forgetting the RfC label.

> Why? ....

No hidden reason. Just the one in the last line of my initial text. :-)
There are two ways to achive the same formatting fot in- and output.
And, if linux is the reference, at least it has s specific error message.

OlegHahm, 2015-11-05:
> And, if linux is the reference, at least it has s specific error message.

How about this: Linux gives this error message for every invalid format of destination addresses, e.g.
```
ping6 foobar
unknown host
```
We could add this line to the error message of shell's ping6 command, if you think that this helps.

A-Paul, 2015-11-05:
@OlegHahm that could be considered "less confusing" too.
I suppose printing addresses without prefix is neither an option.

OlegHahm, 2015-11-05:
> I suppose printing addresses without prefix is neither an option.

Hm, my first reflex was saying: yes, that is out of question, because I'm so used to this representation from using `iproute` tools, a little bit of "research" revealed this might actually a good idea. Even the corresponding [RFC](https://tools.ietf.org/html/rfc5952) doesn't specify the prefix notation for single IPv6 addresses and Linux' equivalent of `ifconfig` also represent the prefix separately (e.g. `net6 fe80::3e97:eff:fe3f:bb02  prefixlen 64  scopeid 0x20<link>`).

authmillenon, 2015-11-05:
>  and Linux' equivalent of ifconfig also represent the prefix separately (e.g. net6 fe80::3e97:eff:fe3f:bb02 prefixlen 64 scopeid 0x20<link>).

@OlegHahm what are you talking about?

```
$ ifconfig
eth0      Link encap:Ethernet  HWaddr 00:16:17:d6:60:c5  
          inet addr:192.168.101.9  Bcast:192.168.101.255  Mask:255.255.255.0
          inet6 addr: fd53:5ee2:328:65:216:17ff:fed6:60c5/64 Scope:Global
          inet6 addr: fe80::216:17ff:fed6:60c5/64 Scope:Link
          inet6 addr: fd53:5ee2:328:10::50a/128 Scope:Global
          inet6 addr: fd53:5ee2:328:65:40bb:26b:29a6:7a71/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1529676 errors:0 dropped:0 overruns:0 frame:0
          TX packets:460082 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2077527933 (2.0 GB)  TX bytes:51937595 (51.9 MB)
```

I'm against removing the prefix length because it would make debugging configuration bugs considerably harder. `fe80::216:17ff:fed6:60c5/64` can have a very different behavior in routing and neighbor discovery than say `fe80::216:17ff:fed6:60c5/80`. That's makes the prefix length important in an interface scope. The prefix length isn't however part of the host's address (and in most cases for remote hosts can't even be assumed to be known) so it seems a bit weird to actually let ping6 accept that. I am however open to add additional error messages for wrong input parameters (might also be helpful as soon as we have DNS in).

A-Paul, 2015-11-05:
Just to gain a little bit of wisdom...

> I'm against removing the prefix length because [...]

Oleg's example has `...b02 prefixlen 64 ...`. Is this also considered as *removing*?
Or did you start raging instantly at the space after the IP? ;)

authmillenon, 2015-11-05:
I'm not raging. I just prefer the [RFC'd method of denoting a prefix](https://tools.ietf.org/html/rfc4291#section-2.3).

OlegHahm, 2015-11-06:
@authmillenon, my ifconfig gives me
```
wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 133.93.24.83  netmask 255.255.248.0  broadcast 133.93.31.255
        inet6 2001:c40:0:3024:a11:96ff:feda:2870  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::a11:96ff:feda:2870  prefixlen 64  scopeid 0x20<link>
```
I'm not voting to remove the prefix, but saying that we could split it if this is helpful.

gebart, 2015-11-06:
ifconfig output is different on different systems, there's no standard among them, and it's been deprecated for many years (see e.g. http://xmodulo.com/linux-tcpip-networking-net-tools-iproute2.html). iproute2 (`ip a`) on the other hand shows the addresses in a consistent way. IIRC RIOT ifconfig is modelled after iproute2.

can't we just make ping6 ignore anything after a slash and be done with this topic?

authmillenon, 2015-11-06:
> ifconfig output is different on different systems, there's no standard among them, and it's been deprecated for many years (see e.g. https://bugzilla.redhat.com/show_bug.cgi?id=784314, http://xmodulo.com/linux-tcpip-networking-net-tools-iproute2.html).

That is only true for Linux (and I'm not even sure if this is true for all distributions).

> iproute2 (ip a) on the other hand shows the addresses in a consistent way. IIRC RIOT ifconfig is modelled after iproute2.

No, it's modeled after (Debian's) `ifconfig`. `iproute2` is far more complex than just interface configuration (as it's name suggest you can manipulate route information and even the neighbor cache with it). That's why we decided to port `ifconfig`, not `iproute2`, to keep the feature set intentionally low (after all its just a debugging tool in the IoT context).

> can't we just make ping6 ignore anything after a slash and be done with this topic?

As already stated: I never was a fan of soft-parsing (to many bad experiences with it), but if others want it than please do.

A-Paul, 2015-11-12:
I used the idea of making the error message more precise in [this PR](https://github.com/RIOT-OS/RIOT/pull/4266)

A-Paul, 2016-01-15:
My already merged change to a more specific error message has reduced the potential confusion regarding the address input.
In between a discussion came up weather to format the prefix as `ADDR/PREFIX` or `ADDR prefixlen PREFIX`.
If there is further need to discuss this i can leave the issue open, otherwise I would close it.

haukepetersen, 2016-03-31:
@gebart, @authmillenon: Anything to add to this issue or are we ready to close it as @A-Paul suggests?

authmillenon, 2016-03-31:
I think that issue is fixed for now. There are differences in Ubuntu's ifconfig (from net-tools 1.60) and Arch's ifconfig (from net-tools 2.10). Since the output thus does not seem to be standardized I say we close this issue, since we then can do our own stuff ;-).

closed 2016-03-31 by authmillenon