RIOT-OS/RIOT #4233: EUI-64 calculation

OlegHahm, 2015-11-07:
I just discovered that [RFC4193](https://tools.ietf.org/html/rfc4193#section-3.2.2) describes an algorithm to generate a Pseudo-Random Global ID from either a MAC address or "a suitably unique identifier, local to the node, should be used (e.g., system serial number)". Maybe we can use this algorithm for creating an EUI-64 from the CPUID?

authmillenon, 2015-11-07:
Looks alright on first side :+1:

OlegHahm, 2016-01-27:
SHA1 is on its way: #4701.

authmillenon, 2016-01-27:
Just to be clear, the algorithm described on the RFC still requires the time of date and also describes the creation of a (somewhat) globally unique *IPv6* address, right?

OlegHahm, 2016-01-27:
Yes, it requires time and date (most of the devices have an RTC and even if not `gettimeofday()` can also be called with xtimer and it doesn't matter if the timestamp is "real"). And it describes how an EUI-64 and an IPv5 address can be obtained.

miri64, 2020-06-30:
@benpicco @maribu this could be interesting for your rework of `luid`.

maribu, 2020-06-30:
> @benpicco @maribu this could be interesting for your rework of `luid`.

I'd say the current `luid` implementation is superior in terms of usability, as

- No NTP timestamp is needed for generation in `luid`
- The same board with the same transceiver setup will always end up with the same local EUI in `luid`

miri64, 2020-06-30:
It _is_ however still problematic when the CPU ID is very similar (see [e.g. last weeks discussion to the users list](https://lists.riot-os.org/pipermail/users/2020-June/001566.html))

benpicco, 2020-06-30:
Is there a small hash algorithm that produces 64 bit output? Hashing the CPU ID would provide better results than just xor in a loop.

maribu, 2022-09-16:
IMO this can be closed.

@benpicco: ping :)