RIOT-OS/RIOT #4234: rpl: initially sending of DIOs is using short intervals

OlegHahm, 2015-11-07:
After joining a DODAG or when initializing a root node, the RPL node will wend out many DIO messages within a short period. This seems to be wrong to me, but maybe I'm missing the intension.

BytesGalore, 2015-11-07:
@OlegHahm what do you mean by many and short period?
Usually the dissemination periods for DIOs are trickle timer based, and starts with the shortest interval when creating a DODAG, i.e. initializing a root node, or joining a DODAG.
(The same applies on local/global repairs)

After some time the dissemination should cease, when the trickle timer rises its interval.

OlegHahm, 2015-11-07:
Sorry, should been more precise and can provide a pcap by tomorrow. The initial interval was below a second which seems odd to me.

cgundogan, 2015-11-09:
@OlegHahm as per [RFC6550](https://tools.ietf.org/html/rfc6550#section-17) the initial trickle parameter for `Imin`  is `2^DEFAULT_DIO_INTERVAL_MIN`. `DEFAULT_DIO_INTERVAL_MIN` is defined as `3` and the unit of `Imin` is `ms`. So the dissemination of `DIOs` in the starting phase is very quick, but slows down quite fast if there are no further inconsistencies detected

gebart, 2015-11-09:
Is 2^3 really a useful default? 8 ms is a very short wait.

cgundogan, 2015-11-09:
@gebart I agree with you, that 8ms might not be the best choice for every use case. Currently this value is not configurable at all (also not at compile time). But assuming a root node would send a configuration object with a different value for DIOIntervalMin, then this value will be used for all nodes throughout the network.
I just did stick with the standard for the default case, though (which doesn't mean that this is the best way to do)

cgundogan, 2015-12-06:
@OlegHahm are you satisfied with the given answers here or should we keep this question open?

OlegHahm, 2015-12-06:
I guess it's okay. Thx.

closed 2015-12-06 by OlegHahm