RIOT-OS/RIOT #4236: rpl: ignoring DIOs without DODAG Configuration option

OlegHahm, 2015-11-08:
According to RFC6550 this option is not required, however in the current implementation a node won't join a until this option is included in the DIO.

cgundogan, 2015-11-09:
@OlegHahm it may not be required, but it makes actually sense to include the dodag configuration in the very first `DIO` upon joining a `DODAG`. One way around this would be to use only default values for all parameters defined in the `DODAG Configuration Option`, but our current implementation makes use of unicast DIS:

>This [DODAG Configuration] option MAY be
   included occasionally by the DODAG root (as determined by the DODAG
   root), and MUST be included in response to a unicast request, e.g. a
   unicast DODAG Information Solicitation (DIS) message.

Assuming a network would not use the default values, how is a node that wants to join be able to decide
which parameters it should use if there is no `DODAG Configuration Option` available in the first `DIO` it sees?
The dodag configuration option is static throughout the whole network and should not change in time.
So just using default values until some DIO contains a different configuration option sporadically seems a little odd to me..

I guess requesting the configuration option is the best way to go here, because you can then decide to **not** include this option for all DIOs, but only when it really matters (when someone wants to join).

OlegHahm, 2015-11-09:
Well, the problem is that (to the best of my knowledge) many implementations do not imeplement DIS messages at all - which means that they are also not handled upon reception. For the [unstrung implementation](https://github.com/mcr/unstrung) it turned out that it neither handles DIS messages nor includes the DODAG Configuration option which lead to the situation where a RIOT node was not able to join a RPL network with an unstrung root. (I don't remember if we tested the other way around.) Joining the network independently from the fact if there is this option included in the DIO seems to have worked fine.

cgundogan, 2015-11-09:
I understand the problem, but IMO DIS messages and configuration options can only be renounced if you operate in a network with given and factual parameters - for all nodes in this network. How else would a node know if a a certain parameter differs from it's default value? We certainly can implement this, but this needs to be configurable and decided before starting to operate as a router (at compile-time maybe, for all nodes, cannot change later). However, to me this feels like hammering on something in the hope of getting things to work. I mean, DIS messages are defined in the RFC and they can be used to strictly ask for configuration options. Making use of them is, personally, the best way to go.

OlegHahm, 2015-11-09:
I agree. However, I'm very much in Jon Postel's famous credo about being conservative on what you send and liberal about what you receive. Hence, dealing with implementations that do not implement all features or message types or may even be non-compliant to the RFC. (At a first glance, I was unable to figure out if sending and handling of DIS messages is mandatory.)

Maybe we can discuss this face-to-face, e.g. tomorrow.

cgundogan, 2015-11-09:
>Hence, dealing with implementations that do not implement all features or message types or may even be non-compliant to the RFC

I completely agree with you on this one.

>(At a first glance, I was unable to figure out if sending and handling of DIS messages is mandatory.)

"it is a MUST to respond with a DIO+Configuration option if a unicast DIS was received". Personally, I interpret this as a transitive requirement to have DIS messages implemented.

>Maybe we can discuss this face-to-face, e.g. tomorrow.

Sure, 8 am? (;

OlegHahm, 2015-11-10:
> >    Maybe we can discuss this face-to-face, e.g. tomorrow.
>
> Sure, 8 am? (;

For breakfast? Don't mess with someone who just returned from a location with large positive time offset. ;)

OlegHahm, 2016-02-27:
@cgundogan, let's discuss next Tuesday.

PR #5082 (2016-03-21): gnrc_rpl: make DODAG_CONF optional when joining DODAG
Fixes #4236 (see discussion there).

@OlegHahm 
>@cgundogan, let's discuss next Tuesday.

We never discussed it, but I am fine providing that feature as a compile-time option.
closed 2016-03-21 by cgundogan