RIOT-OS/RIOT #4243: Add emul8 embedded hardware emulator support

gebart, 2015-11-10:
http://emul8.org, https://github.com/emul8/emul8

Contiki-developers mailing list announcement: http://sourceforge.net/p/contiki/mailman/message/34606321/

Emul8 is an emulator for embedded hardware. We could use it to run tests in the CI system and during development.

OlegHahm, 2015-11-10:
And for the ARCH Linux users: see the issue tracker of the emul8/emul8#1 project.

kaspar030, 2015-11-10:
Very promising! +1

mgielda, 2015-11-16:
Hi guys, glad to see the interest! As you know we would be very happy to do this. We also want to integrate with the Contiki test suite, so this is only natural to have this in place for RIOT too.

mgielda, 2015-11-16:
By the way, I had been running RIOT for Zynq and cc2538 binaries in Emul8 no problem, let me create a sample project for you. Note that any APIs etc are subject to change in the nearest future, as we will be working on the user experience and integrability of Emul8.

OlegHahm, 2015-11-16:
> let me create a sample project for you

Cool! Looking forward to this!

mgielda, 2015-11-19:
https://github.com/antmicro/emul8-sample-riot-project - here you go :) sorry for the delay, so much to do!

OlegHahm, 2015-11-19:
No problem at all - I know what you're talking about. ;)

miri64, 2016-10-18:
Can we somehow get this into the main repository? This is not very prominent.

miri64, 2016-11-02:
Postponed.

basilfx, 2017-11-06:
I've created #7945 based on the example of @mgielda.