RIOT-OS/RIOT #4246: rfc: UART hardware flow control support

DipSwitch, 2015-11-10:
I'm currently busy with with implementing an full functioning UART and I propose to add the defines `UART_#_CTS_PIN` and `UART_#_RTS_PIN` to the `perhip_conf.h` files if you want to enable hardware flow support for the given uart by default and additionally add the function `void uart_flowcontrol(uart_t uart, bool rts, bool cts);` in `includes/periph/uart.h` (which have to be implemented in the cpu peripheral drivers to enable or disable them at runtime.

What do you think about this?

haukepetersen, 2015-11-11:
So far we left the flow control configuration to be handled by a specific implementation (e.g. see the `pca10000` board using the `nrf51822` cpu). In my experience, hardware flow control is something you don't see too often, and this is why I didn't include it into the periph driver API. 

Using flow control or not is dependent on the hardware design (e.g. board), so I still think it makes sense to configure this in the board's periph_conf instead of configuring it during runtime using some API function.

DipSwitch, 2015-11-21:
Oke then we leave it to that for :)

haukepetersen, 2015-11-23:
cool. So is it ok for you then to close this issue? We can always re-open it if there are new insights...

closed 2015-12-13 by DipSwitch