RIOT-OS/RIOT #4254: toolchain directory

haukepetersen, 2015-11-11:
Just stumbled once more over the `RIOTDIR/toochain` folder. Does anyone use this and/or is the included toolchain tested and up to date? 

I'd like to propose the following: Let's remove this folder (together with the compile scripts in `dist/tools/toolchains/` from RIOT master and dependent if they are still used either move them to a separate toolchain repo or drop support for them.

Any opinions on this?

OlegHahm, 2015-11-11:
Sounds good to me.

gebart, 2015-11-11:
It has something to do with the qemu-i386 platform, I don't remember the details.

OlegHahm, 2015-11-11:
That's why I assigned [Mr. x86](https://github.com/kijewski)

Kijewski, 2015-11-11:
Used in the precompiled newlib: [boards/x86-multiboot-common/Makefile.include](https://github.com/RIOT-OS/RIOT/blob/7b10e59ef3ef37cde032ddcc4c25b74f781d7428/boards/x86-multiboot-common/Makefile.include).

haukepetersen, 2015-11-18:
I see, in this case I think it makes sense to keep it. I consider this solved -> closing the issue.

closed 2015-11-18 by haukepetersen