RIOT-OS/RIOT #4260: Create NTP client

gebart, 2015-11-12:
NTP is widely used for synchronizing clocks over the Internet. Many IoT applications need a valid system time in order to coordinate with other nodes and to organize data.

An NTP client needs to be able to precisely measure and adjust the system time in order to achieve a good synchronization. It may also be worthwhile to investigate gettimeofday etc.

OlegHahm, 2015-11-12:
I think NTP is a poor choice for constrained node networks that operate mostly over low-power wireless links. A lof of work has been done in this domain for designing more efficient (and most of the time more precise) protocols, such as gtsp or pulsesynch. See also: https://github.com/RIOT-OS/RIOT/pull/1538

miri64, 2016-10-18:
An [SNTP](https://tools.ietf.org/html/rfc5905#section-14) implementation is our project for the [Ladies that FOSS hackathon](https://www.wikimedia.de/wiki/Ladies_that_FOSS)

miri64, 2016-11-02:
Fixed ;-)

miri64, 2016-11-02:
(see #6020)

closed 2016-11-02 by miri64