RIOT-OS/RIOT #4263: Performance of random numbers

PeterKietzmann, 2015-11-12:
I don't know if this is known behaviour, neither I looked at the algorithm.  This is just reporting! The computation time of [`genrand_uint32`](https://github.com/RIOT-OS/RIOT/blob/master/sys/random/mersenne.c#L121) has peaks sometimes. While it takes mostly about some us, one or two of 1000 computations take like hundreds (~300) of us. As @kaspar030 presented  an alternative in #4079, I assign him.

Please note that the code is not developed especially for RIOT. Feel free to close the issue!

gebart, 2015-11-12:
Mersenne twister will have to perform some more computations once every 625 iterations IIRC.

PeterKietzmann, 2015-11-17:
Seems to be a known issue. Alternative is on its way.

OlegHahm, 2015-11-17:
At least for me that was an unknown fact and since @kaspar030's PR is not yet merged I would prefer to keep this open.

PeterKietzmann, 2015-11-27:
It is and `minstd` acts quite efficient

closed 2015-11-27 by PeterKietzmann