RIOT-OS/RIOT #4269: few examples

macknill, 2015-11-13:
Hello. I have two boards SAMR21-XPRO, they ran the project "Hello world." But I do not understand what I do to get them to communicate with each other. Loading various examples, trying to ping across the shell, but to no avail.
I think that the project needed more examples. I am hope for your help.

bapclenet, 2015-11-13:
Have you tried the example posix-socket [0] or gnrc-networking [1]?
If you follow the instruction you should be able to send data form a board to another easily

[[0] https://github.com/RIOT-OS/RIOT/tree/master/examples/posix_sockets](https://github.com/RIOT-OS/RIOT/tree/master/examples/posix_sockets)
[[1] https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking)

PeterKietzmann, 2015-11-16:
@macknill, welcome to RIOT! As @bapclenet already indicated, there are examples that do exactly what you're asking for. In the regarding folders you'll also find short introductions (README). I'd like to close this issue because IMO it is "fixed". If you have further questions, please state them on the developers or users mailing list. Fine with that?

closed 2015-11-16 by PeterKietzmann

PeterKietzmann, 2015-11-16:
In addition: I have the feeling that there is currently some bug hunting going on at the master branch. You might want to checkout the release branch [2015.09 branch](https://github.com/RIOT-OS/RIOT/tree/2015.09-branch)