RIOT-OS/RIOT #4271: auto_init: refined concept for driver initialization

haukepetersen, 2015-11-13:
With the `gnrc` stack we introduced a new concept of initializing network interfaces, which is based on the following: 
- each network device driver defines a struct with the parameters it needs for initialization (naming scheme is `xx_params_t`
- for a device configuration you have to define a header file named `xx_params.h`, which in turn instantiates one ore more (const) instances of this struct containing the configuration wanted (this allows for multiple instances of a device...)
- you can define this configuration for a board (`boards/BOARDNAME/include/xx_params.h`) or specific for an application. 
- by putting you application folder in the include path before the boards include folder, you can override any board specific configuration with your application specific one
- `auto_init` includes one 'bootstrap' file per device, which implements the init function `auto_init_xx()` and allocates memory for stacks, devices descriptors or whatever is needed

Seems to me like this concept has proven to work just fine - so I'd like to make some suggestions on how to generalize this one step further for all device drivers to work:
- each device driver that uses peripherals MUST define a `xx_params_t` struct
- each device driver MUST use this type as argument for its init/setup function
- each device driver MUST provide a default `xx_params.h` file, which sets the neede parameters to default values while allowing manual override (e.g.  `CFLAGS=-DXX_PARAM_SPI=SPI_7 make all`)

This scheme would allow for more compact board configurations while still allowing the flexibility of overriding configurations and defining periph mappings per Makefile/commandline.

See #4270 for an example for the Atmel radio driver.
#3538 contains some preparation for sensors and actuator devices.

haukepetersen, 2015-11-13:
Oh, and this would also enable easy integration of hardware that is connected to boards through pre-defined hardware interfaces (plugs, defined extension headers, arudino-shields). 

For example the Atmel Ethernet1-Xpro: it uses the same pin definitions (though a common extension header) for multiple Atmel boards (`samr21-xpro`, `samd21-xpro`). I would be enough to define a global header with this configuration which just needs to be include by the `board.h` files of both boards - and voila - once you include the driver and use one of these boards, you automaticall end up with the right peripheral configuration for the driver...

kaspar030, 2015-11-28:
+1

You tagged "Release 2015.12", did you intend to convert all drivers until then (until next week)?

haukepetersen, 2015-11-30:
nope :-)

immesys, 2017-02-22:
@haukepetersen in this system, how does an application get access to the device handle initialized by the auto_init_xx() function?

haukepetersen, 2017-11-02:
The concept above is basically what we do today. It still has flaws (as pointed out by @immesys above), and we definitely need to do another major iteration on our overall device driver model concepts. But for now I consider this issue closed.

closed 2017-11-02 by haukepetersen