RIOT-OS/RIOT #4275: auto_init_netif_* modules always use the 'nomac' module

rousselk, 2015-11-15:
When using the `auto_init_gnrc_netif` module, this one initialize the radio transceiver, but also automatically starts the `gnrc_nomac` module.

This is quite annoying for testing new MAC protocols, since you have to initialize the network machinery in your own application to be able to use your own MAC protocol module. (See the test applications I made in PR #4180 and #4184.)

A way to configure which MAC layer to initialize with `auto_init_gnrc_netif` and use would be a very useful addition.

Unfortunately, I won't have the time to design and implement such a mechanism now.

haukepetersen, 2015-11-16:
Agreed, the mac layer should be configurable through auto_init. I think we should be able to find a concept here...

haukepetersen, 2016-03-31:
postponed as we won't solve this for this release

kYc0o, 2016-07-26:
and postponed again... maybe @zhuoshuguo or @miri64 can have an idea?

miri64, 2019-08-10:
`gnrc_nomac` isn't a thing anymore

closed 2019-08-10 by miri64