RIOT-OS/RIOT #4276: dist/tools/eclipsesym: not working under OS X

basilfx, 2015-11-15:
Hi,

Using `make eclipsesym` doesn't work under OS X. It outputs the following:

```
printf "%s\n" arm-none-eabi-gcc -DDEVELHELP -Werror -DMODULE_AUTO_INIT -DMODULE_CORE -DMODULE_CORTEXM_COMMON -DMODULE_CPU -DMODULE_EMLIB -DMODULE_NEWLIB -DMODULE_PERIPH -DMODULE_PERIPH_COMMON -DMODULE_SYS -DMODULE_TSRB -DMODULE_UART_STDIO -mcpu=cortex-m3 -mlittle-endian -mthumb -mfloat-abi=soft -mno-thumb-interwork -std=gnu99 -Wall -Wstrict-prototypes -Werror=implicit-function-declaration -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -ggdb -g3 -Os -DCOREIF_NG=1 -DCPU_MODEL_EFM32LG990F256 -DCPU_ARCH_CORTEX_M3 -DBOARD_STK3600=\"stk3600\" -DRIOT_BOARD=BOARD_STK3600 -DCPU_EFM32LG=\"efm32lg\" -DRIOT_CPU=CPU_EFM32LG -DMCU_EFM32LG=\"efm32lg\" -DRIOT_MCU=MCU_EFM32LG -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -DRIOT_VERSION=\"2015.12-devel-504-gd108-MacBook-van-Bas.lan\" -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/core/include -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/drivers/include -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/sys/include -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/boards/stk3600/include -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/cpu/efm32lg/emlib/inc -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/cpu/efm32lg/include -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/cpu/cortexm_common/include  -I/Users/basilfx/SimplicityStudio/v3_workspace/lib/sys/libc/include  | \
		/Users/basilfx/SimplicityStudio/v3_workspace/lib/dist/tools/eclipsesym/cmdline2xml.sh > /Users/basilfx/SimplicityStudio/v3_workspace/src/eclipsesym.xml
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
readlink: illegal option -- m
usage: readlink [-n] [file ...]
```

I believe it has something to do with `readlink` not being compatible with GNU `readlink`. I have homebrew installed, which installs `greadlink`. If I manually change `readlink` to `greadlink`, the problem is solved for my machine.

Using up-to-date `master` branch with OS X 10.11.

Lotterleben, 2015-11-22:
Ahhh, sorry for the late response :( I'm afraid I'm useless here since I use a VM for RIOT, but maybe other mac users (@josephnoir , @emmanuelsearch , @thomaseichinger?) have an idea?

thomaseichinger, 2015-11-23:
@basilfx Could you please check, if the generated file also works with omitted `-m` flag?

PR #4330 (2015-12-08): eclipsesym/cmdline2xml.sh: introduce workaround readlink function
readlink for OS X and FreeBSD does not offer -m flag to canonicalise the
file's path. Implement a workaround shell function to simulate behaviour.

Fixes #4276
closed 2015-12-08 by cgundogan