RIOT-OS/RIOT #4281: core/lpm_arch: Implement peripheral shutdown and startup

DipSwitch, 2015-11-16:
It would be nice to have a hook functions from the `_idle_thread` so you can enable / disable peripherals and/or devices before going to sleep. This you would want to overload on the board or application.

Thoughts about this idea?

kaspar030, 2015-11-28:
Hmm, maybe for the deep-sleep states, otherwise, whenever a thread sleeps (so the scheduler switches to idle), the hooks would need to be executed.

punchcard60, 2015-12-13:
Having the ability for "applications" and device drivers to power devices on and off when needed is an integral part of power management. I would like to propose that this issue be combined with #2927 as part of Active Power Management. [This wiki page](https://github.com/RIOT-OS/RIOT/wiki/Active-Power-Management-Task-Force) has more information.

lebrush, 2017-03-03:
@haukepetersen @kaspar030 is this outdated after the PM implementation?

kaspar030, 2017-03-03:
The concept of the new power management would be that the applications themselves turn off devices when they don't need them anymore, which in turn would make the device unblock sleep modes, which then would make the idle thread select the deepest possible sleep mode.

Would that serve your need?

lebrush, 2017-03-03:
Mine yes :-) @DipSwitch ?

PeterKietzmann, 2017-03-28:
I guess it's outdated. Will close, @DipSwitch please reopen if I'm mistaken.

closed 2017-03-28 by PeterKietzmann