RIOT-OS/RIOT #4291: shell: shell commands for random module(s)?

haukepetersen, 2015-11-17:
Just has this thought when looking through the code of #4287:

Why again has the random module(s) its/their own shell command? Having them in a test app might make sense, but having these (on its own rather useless) shell commands in all apps that use shell commands seems like wasted memory to me... 

I would tend to move the shell commands to a specific test application!

kaspar030, 2015-11-17:
totally +1.

kaspar030, 2015-11-17:
But assigning me - so you want me to do this? :)

authmillenon, 2015-11-17:
I talked with @mehlis about this way back. The reason is, that there is no seed-initialization currently. #3425 would remove the need for the shell commands.

kaspar030, 2015-11-17:
We should come up with a way to use random events (button presses, incoming network replies, ...) to seed our randomness...

OlegHahm, 2015-11-17:
You're talking about https://github.com/RIOT-OS/RIOT/issues/3221?

authmillenon, 2015-11-17:
That's why I started #3425 do give a starting point for that (hardware-randomness if available, xtimer + cpuid if available, etc.). It's better than nothing (or a shell command, that only can be called after start-up).

OlegHahm, 2015-11-17:
Well, actually, I remember that we discussed this question already a couple of times back in good ol' µkleos days - but always without finding a good answer to the question how to determining a good seed.

kaspar030, 2015-11-17:
@OlegHahm I was thinking of something like #4294.

haukepetersen, 2015-11-18:
ok, initializing the seed via shell is something I did not think of, so there we have a right to exist for the shell command - that was the answer I was looking for. 

How about to close this issue for now and remove the shell command once we have agreed on a 'seed'-concept?

kaspar030, 2015-11-28:
Yeah, one more solved issue is a good thing. :)

closed 2015-11-28 by kaspar030