RIOT-OS/RIOT #4309: documentation: selected topics

OlegHahm, 2015-11-19:
- [ ] Create an application
 - [ ] Networking
 - [ ] The `main()` funciton
 - [ ] Makesystem
 - [ ] Include modules
 - [ ] Threading
 - [ ] Choose the right stack size
 - [ ] IPC
 - [ ] Auto initialization
 - [ ] scheduler
 - [ ] interrupts
 - [ ] porting guide (+ periph API ?)
 - [ ] driver model
 - [ ] packages

emmanuelsearch, 2015-11-26:
Extract of email received: 
``` 
I have not found a good documentation on ISR and Scheduler for RIOT.
I need to get a bit deeper to understand if ISR is executed in current task context and checking if task switch is required. 
Appreciate if you could point me to a good document.
```

authmillenon, 2015-12-14:
Structured the points a bit.

OlegHahm, 2015-12-15:
What about xtimers? I guess that's another important module that needs some high-level description.

authmillenon, 2015-12-15:
Yapp. Feel free to add ;-)

cgundogan, 2016-01-07:
This issue does not seem to reflect the current state of documentational work done. What's the purpose of keeping this issue open?

authmillenon, 2016-01-07:
Update it if its not up-to-date. I still think the doc needs a lot of work ;-)

haukepetersen, 2016-01-21:
Yep. I am still heavily working on doc for cpu/boards/drivers.

Lotterleben, 2016-01-21:
Dumb question: What's the criteria for 'selected'? :)

OlegHahm, 2016-01-22:
That someone(TM) selected them. (*scnr*) More seriously: we tried to identified the most important high level concepts a RIOT newcomer needs to grasp.

miri64, 2016-10-18:
There is some overlap to https://github.com/RIOT-OS/RIOT/issues/1837. Can we somehow merge them?

maribu, 2023-05-17:
I like the idea of a tracking issue on improving doc. I think we still have areas left that could need a lot of improvement.

I think that the tracking issue here however is a bit outdated with some areas (xtimer, conn) no longer being relevant.