RIOT-OS/RIOT #4342: [RFC] thread-decoupled messaging

authmillenon, 2015-11-25:
https://github.com/RIOT-OS/RIOT/pull/4327 and some discussions we had in the last few days showed, that we really are in the need of some alternative (or completely overhauled) IPC, that does not rely on threads and maybe has some kind of prioritization, too. This issue is here to discuss the general specification of such architecture (because doing so proofed to be a success in with GNRC ;-)):

I was thinking about something in the line of a centralized, thread-safe, priority queue for messages

* Data type: message queue… let's call it message box (`mbox_t`) to not confuse it with the threaded message queues.
* Operations:
  * `int mbox_put(mbox_t *mbox, msg_t *msg, int prio)` (equivalent to `msg_send()`)
  * `int mbox_try_put(mbox_t *mbox, msg_t *msg, int prio)` (equivalent to `msg_try_send()`)
  * `int mbox_get(mbox_t *mbox, msg_t *msg)` (equivalent to `msg_receive()`)
  * `int mbox_try_get(mbox_t *mbox, msg_t *msg)` (equivalent to `msg_try_receive()`)
* optional operations (e. g. through a submodule)
  * `int mbox_put_timed(mbox_t *mbox, msg_t *msg, int prio, xtimer_t *timer, uint32_t offset)` equivalent to `xtimer_set_msg()`
  * `int mbox_put_timed64(mbox_t *mbox, msg_t *msg, int prio, xtimer_t *timer, uint54_t offset)` equivalent to `xtimer_set_msg64()`
  * `int mbox_get_timeout(mbox_t *mbox, msg_t *msg, uint32_t timeout)` equivalent to `xtimer_msg_receive_timeout()`
  * `int mbox_get_timeout(mbox_t *mbox, msg_t *msg, uint64_t timeout)` equivalent to `xtimer_msg_receive_timeout64()`

I'm not sure if equivalents for  `msg_send_receive()` and `msg_reply()` are necessary (probably not), but they can be added as wrappers for `mbox_put()` and `mbox_get()` respectively, I guess.

authmillenon, 2015-11-25:
As an implementation note: I think an array-based heap-structure could be sufficiently fast and light-weight.

kaspar030, 2015-11-25:
As I said, I have code for this ready. interface looks almost the same. I'll dig it out.

authmillenon, 2015-11-25:
Oh nice! Did not heard it yesterday :(

haukepetersen, 2016-03-31:
@kaspar030, @authmillenon: what's the state on this. Did the thread flags give us any advantage in this direction? And are we gonna solve this for 2016.04?

authmillenon, 2016-03-31:
This is fixed by #4919. Since one is postponed to 2016.07 I say we do the same for this issue.

kYc0o, 2016-07-26:
The PR solving this was merged, so I'll close the issue.

closed 2016-07-26 by kYc0o