RIOT-OS/RIOT #4344: openwsn make file

IsidurPaine, 2015-11-26:
Hi everyone,
i'm making my way round openwsn & RiOT.
As I try to 

make -B clean flash

in

/home/piero/Scrivania/RIOT/pkg/openwsn

I get this:

make: Warning: File `Makefile' has modification time 1,3e+03 s in the future
Cleaning up OpenWSN package...
Removing Makefile~
Resolve operation not in progress, we are not resuming.
HEAD is now at ff25e5d Merge pull request #125 from NicolaAccettura/develop_FW-309
make[1]: Entering directory `/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn'
make[1]: *** No rule to make target `patch'.  Stop.
make[1]: Leaving directory `/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn'
make: *** No rule to make target `flash'.  Stop.

Could someone help?
Thanks

IsidurPaine, 2015-11-26:
Somehow, it got one step forward on its own.
As i'm in /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn and 

make

it tries to (line 18)

include $(RIOTBASE)/Makefile.base

and the output is

Applying: Add RIOT Makefiles
Applying: Add RIOT adaption
Applying: Fix old style definitions
Applying: provide openserial dummies
Applying: fixes to RIOT adaption
"make" -C /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn
make[1]: Entering directory `/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn'
Makefile:18: /Makefile.base: No such file or directory
make[1]: *** No rule to make target `/Makefile.base'.  Stop.
make[1]: Leaving directory `/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn'
make: *** [all] Error 2

Now the question is changed: Makefile.base is located in RIOT main folder (/home/piero/Scrivania/RIOT) and this Makefile is downloaded every time i call make in pkg/openwsn so it looks like it's not possible to substitute "RIOTBASE" with another path.
Is there anything I can do to solve this?

thomaseichinger, 2015-11-26:
@IsidurPaine The path you are in, represents the package only. It contains a Makefile and patches to fetch and adapt OpenWSN. To actually build an application using this package please take a look here https://github.com/RIOT-OS/applications/tree/master/openwsn
Please keep in mind though, this mentioned application was not adapted to recent module name changes. Neither was the OpenWSN pkg yet. Though there is work in progress to provide deeper 6TiSCH support in RIOT, by integrating OpenWSN's MAC layer implementation in RIOT.

IsidurPaine, 2015-11-26:
Thank you @thomaseichinger.
Now I'm in /Scrivania/RIOT-applications/openwsn and as I make (telosb are mandatory for me) i got this

Building application "openwsn-app" for "telosb" with MCU "msp430fxyz".
"make" -C /home/piero/Scrivania/RIOT/pkg/openwsn
"make" -C /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn
"make" -C /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn/bsp/boards/riot-adaption
In file included from /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn/bsp/boards/board_ow.h:16:0,
                 from /home/piero/Scrivania/RIOT/pkg/openwsn/openwsn/bsp/boards/riot-adaption/board_ow.c:1:
/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn/bsp/boards/toolchain_defs.h:25:0: error: "ISR" redefined [-Werror]
/home/piero/Scrivania/RIOT/cpu/msp430-common/include/cpu.h:45:0: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[3]: *** [/home/piero/Scrivania/RIOT-applications/openwsn/bin/telosb/openwsn/board_ow.o] Error 1
make[2]: *** [ALL--/home/piero/Scrivania/RIOT/pkg/openwsn/openwsn/bsp/boards/riot-adaption] Error 2
make[1]: *** [all] Error 2
make: *** [/home/piero/Scrivania/RIOT-applications/openwsn/bin/telosb/openwsn.a] Error 2

The ISR redefinition depends on the 2 different declarations in toolchain_defs.h and cpu.h, which respectively are:
#if defined(__GNUC__) && (__GNUC__==4)  && (__GNUC_MINOR__<=5) && defined(__MSP430__)
[...]
   #define              ISR(v) interrupt (v ## _VECTOR) v ## _ISR(void)
#else
   [...]
   #define              ISR(v) __PRAGMA__(vector=v ##_VECTOR) __interrupt void v ##_ISR(void)
#endif

and 

#define ISR(a,b)        void __attribute__((naked, interrupt (a))) b(void)

Is there anything I can do?
Thanks.

thomaseichinger, 2015-11-27:
@IsidurPaine The porting of OpenWSN in this package is not very complete. It was not tested on other than ARM platforms and also depends on a Atmel at86rf2xx transceiver by now. (Mostly because the network driver API was not capable to satisfy all of OpenWSN's need at that time.)

That said and as mentioned before, this will improve significantly in the next months as we take an other approach and integrate OpenWSN's MAC features in RIOT directly. This way we want to provide support for any of RIOT's supported radios and easier integration in RIOT applications.

Out of curiosity, which radio module do you intend to work with?

IsidurPaine, 2015-12-01:
Sorry, my answer came late. As said, telosb are mandatory so i'm using cc2420.

haukepetersen, 2016-03-31:
@thomaseichinger, @IsidurPaine: does this issue still exist?

IsidurPaine, 2016-03-31:
Hi,
as far as I've been notified, situation is still the same.
I'm looking forward.

haukepetersen, 2016-03-31:
ok, @thomaseichinger any chance this issue will be fixed for this release?

thomaseichinger, 2016-04-02:
@haukepetersen @IsidurPaine This PR will need to be updated because of the port of at86rf2xx to netdev2. Beside this for TelosB support we'd need to pull cc2420 driver to netdev2 too. 
I can work on parts of it but I won't expect this issue will be solved for the 2016.04 release.

kYc0o, 2016-07-26:
The openwsn package is broken, let's open another issue to point it out and work on this one for the next release.

miri64, 2016-10-17:
@kYc0o did this happen?

kYc0o, 2016-10-20:
Nein, not yet, I'll try to test it soon.

miri64, 2016-11-02:
@kYc0o well can you do this? I'll postpone this one and set a memo label for you to remember ;-)

kYc0o, 2016-11-02:
Sorry I have it in my TODO list but didn't reach it... I have spotted it while doing the tests so I think I'll spot it again ;)

OlegHahm, 2016-11-02:
I took a look and I think it would really require some more work to fix this. Hence, I would advice to temporarily drop support for OpenWSN in RIOT. :cry:

PeterKietzmann, 2017-01-26:
We dropped it #6359. Will close with memo label set in case someone will work on a new port sometime.

closed 2017-01-26 by PeterKietzmann