RIOT-OS/RIOT #4356: boards/telosb: serial input loses characters

thomaseichinger, 2015-11-27:
Connecting using `make term BOARD=telosb`: 
```bash
2015-11-27 16:23:56,391 - INFO # > RIOT MSP430 hardware initialization complete.
2015-11-27 16:23:56,391 - INFO #
2015-11-27 16:23:56,413 - INFO # main(): This is RIOT! (Version: 2015.12-devel-628-g1f7829-et-mbpr.lan-pr/telob_tty)
2015-11-27 16:23:56,413 - INFO # Welcome to RIOT!
help
2015-11-27 16:24:04,665 - INFO # > hl
2015-11-27 16:24:04,681 - INFO # shell: command not found: hl
ps
2015-11-27 16:24:08,791 - INFO # > p
2015-11-27 16:24:08,792 - INFO # shell: command not found: p
helphelphelp
2015-11-27 16:26:23,789 - INFO # > hlhelphelp
2015-11-27 16:26:23,791 - INFO # shell: command not found: hlhelphelp
```
It seems to lose 2nd and 3rd character only...

haukepetersen, 2015-11-27:
probably caused through missed uart interrupts - I had once an (ugly) hack for the msp430 boards in place, that prevented this, but I think @kaspar030 reworked that part, right?

haukepetersen, 2016-03-31:
the telosb is running at ~2MHz main clock, I guess this is not enough to process the UART at 115200 baud.  -> #5215

PR #5215 (2016-04-01): boards/telosb: changed default baudrate to 9600
supposedly fixes #4356

I don't have a board to test here, but I assume this should fix the issue of dropped characters on the `telosb`, as the main clock is currently configured with ~2MHz, and this is not enough to process UART interrupts at 115200baud...
closed 2016-04-01 by cgundogan