RIOT-OS/RIOT #4363: Uncrustify all files for release?

authmillenon, 2015-11-29:
Should we run uncrustify over the whole repo before the release? This would have the advantage, that we can make style checks in CI afterwards.

kaspar030, 2015-11-29:
Not sure I like all of the uncrustify settings. E.g., it uses tabs to align static array initializers.

gebart, 2015-11-29:
Let's fix those settings first so it doesn't mess up the code style.

OlegHahm, 2015-11-29:
Actual, I'm against major code formatting changes. Most of the code is in good shape anyway and it will just screw up git blame results.

x3ro, 2015-12-01:
@authmillenon does have a point though. We'd have to do this at some point if we ever want to do automated style checks. If we manage to configure uncrustify such that it mostly matches the style of the current code, the changes might be relatively few, right? 

I'm not advocating this for the next release, but maybe we should check the impact of running uncrustify over all the source and then decide on whether or not automated style checks are worth it :)

OlegHahm, 2015-12-01:
>  if we ever want to do automated style checks

I'm still not convinced if this is a reasonable goal. There are many good reasons to deviate from style conventions.

Nevertheless, I won't object if someone wants to spend some time in improving uncrustify's config (or proposes a different tool).

OlegHahm, 2015-12-01:
I removed the milestone, because I consider the probably very huge changeset for this proposal as undoable within the given timeframe. Also, there is no urgency here.

haukepetersen, 2016-03-31:
if I remember the discussion in #4792 correctly, there were not too many friends (me included) in just running uncrustify over `core`, as there are some places, where manual fomatting just yield better readable results. So can we take the outcome of that discussion and apply it to this issue also?

authmillenon, 2016-03-31:
I would say yes, but maybe it is a good idea to look over the code regardless. I think some issues are still in core.

haukepetersen, 2016-03-31:
> I think some issues are still in core.

I completely agree, but I am just not a big fan of simply trusting automated tools to do the job better than a manual review :-)

authmillenon, 2016-03-31:
Dito :D. But doing it and looking at the output is still something we could do for a release.

A-Paul, 2016-03-31:
@haukepetersen, they can do better at certain tasks because tools don't get tired/bored or lack concentration. But their use should be *well* prepared. So, I agree that "simply trusting automated tools" isn't a good idea.

Before producing such a massive impact to the whole code base, if ever, we have to have an almost complete and unambiguous set of rules all can stick to. That is the part a tool can't solve for you.

Style check maybe nice at some point but it is not essential. So, no hurry.

haukepetersen, 2016-04-01:
> we have to have an almost complete and unambiguous set of rules all can stick to

Yepp, and that is the problem :-)

But I agree, let's work toward this, but not for this release.

kYc0o, 2016-07-26:
And not for this one either... :/ hopefully the next?

jia200x, 2018-01-29:
> if I remember the discussion in #4792 correctly, there were not too many friends (me included) in just running uncrustify over core, as there are some places, where manual fomatting just yield better readable results. So can we take the outcome of that discussion and apply it to this issue also?

Maybe a dist/tools/uncristify command with several blacklist entries? (e.g core or specific files that won't need uncrustify)
Since most reviews are style issues, running uncrustify on most PRs could help to reduce reviewing time and merge faster.

miri64, 2019-09-10:
See https://github.com/RIOT-OS/RIOT/pull/8519#issuecomment-529812982

closed 2020-11-16 by stale[bot]