RIOT-OS/RIOT #4370: core/kernel_init: possibility to increase main stacksize

PeterKietzmann, 2015-12-01:
The stack size of the main function (`THREAD_STACKSIZE_MAIN`) currently depends on the default thread stack size (`THREAD_STACKSIZE_DEFAULT`) plus a printf-offset (`THREAD_EXTRA_STACKSIZE_PRINTF`) defined by the cpu. `THREAD_STACKSIZE_MAIN` is not just used for initializing the main thread, but also in other locations. Does anyone see the need for increasing **just** the stack size of the main thread?

Feel free to close if this issue doesn't find approval any time soon.

OlegHahm, 2015-12-01:
I'm not sure if I get the question/proposal. You want to increase the stacksize for the main thread, but not all other threads that use the same macro?

PeterKietzmann, 2015-12-01:
Jap, that is the question.

PeterKietzmann, 2015-12-15:
It seems there is no need for such spacial feature. Therefore I'll close this one again

closed 2015-12-15 by PeterKietzmann