RIOT-OS/RIOT #4371: periph/timer: API inconstency

OlegHahm, 2015-12-01:
Reviewing #4089 I realized that `periph/timer` API seems to be inconsistent. `timer_init()` returns 0 on success, while all other functions return 1 on success. I think all periph functions which just return a boolean value should return the same one. Ideally this would be rather a macro like `PERIPH_SUCCESS` than a literal.

dkm, 2015-12-02:
same kind of inconsistency exists in nrf24l01p driver : some functions return 0 (preload, unregister, ...) on success, some return 1 (set_channel, ...).

PeterKietzmann, 2015-12-02:
@dkm you're right. As stated in #4389 I think the whole driver could need some rework. But it's not part of this issue :-)

haukepetersen, 2016-03-31:
all driver interfaces except `spi` and `i2c` have been reworked, and including these two they all have the same return values.

@OlegHahm: is this enough to close this PR, or do you insist on some macro on top of this?

A-Paul, 2016-03-31:
It may be a surprise, ;-) but I still recommend using macros. Not sure if there is need to list all the advantages...
I might help finding names and applying them.
Preferably after this release.

haukepetersen, 2016-04-01:
I don't care (though I am not a fan of it). So lets keep this issue open.

OlegHahm, 2016-04-01:
1. I don't see the initial issue solved. `timer_init()` still returns 0 on success, while the other functions return 1 on success.
2. I agree with @A-Paul here, that macros would be preferable (and would help to avoid such a confusion).

kYc0o, 2016-07-26:
Still inconsistent. Let's move it for the next release.

miri64, 2016-11-02:
This somehow flew under my radar. Postponed.

closed 2019-09-10 by stale[bot]