RIOT-OS/RIOT #4375: cc110x: cc110x generates IID from CPUID, not from hardware address

authmillenon, 2015-12-01:
The IID should be generated from a netdev's hardware address (or one of it, depending on the device). However, the cc110x driver currently generates it from the CPUID for some reason. In the cc110x's case it should be `00:00:00:ff:fe:00:00:xx` where `xx` is the device's hardware address.

OlegHahm, 2015-12-01:
Just checking: you're aware that the cc110x device doesn't store any persistent hardware address?

kaspar030, 2015-12-03:
@authmillenon *ping*

The alternative to cpuid is using a random value, which is not always as unique as it should.

authmillenon, 2015-12-03:
> Just checking: you're aware that the cc110x device doesn't store any persistent hardware address?

yes, but the IID should be based on it.

> @authmillenon ping
>
> The alternative to cpuid is using a random value, which is not always as unique as it should.

The problem isn't with the cpuid (the opposite is the case IMHO the initial hardware address should be generated from the cpuid, as with the other devices), the problem is, that the IID is not based on the hardware address as described in [RFC 4944](https://tools.ietf.org/html/rfc4944) (with PAN ID being padded) or as we had it in the old stack. See also [the 6LoBAC draft](https://tools.ietf.org/html/draft-ietf-6lo-6lobac-03#section-6) for generation of IIDs from an 8-bit hardware address.

authmillenon, 2015-12-03:
Just to be clear: this is a problem, because for Link-layer addresses 6Lo-ND uses back-translation of the IID to the hardware address, but when it has nothing to do with it, 6Lo-ND will not work.

authmillenon, 2015-12-03:
That's maybe why the user in [this users ML thread](https://lists.riot-os.org/pipermail/users/2015-November/000463.html) had problems running 6LoWPAN on cc110x.

PR #5180 (2016-03-26): cc110x: cc110x generates IID from CPUID, not from hardware address
Fixes #4375 and #4998 and makes cc110x more similar to 2-bit address mode of IEEE 802.15.4 (which means better compression with 6LoWPAN). Cases of duplicate addresses can be fixed in a seperate issue by fixing #208 / #3052
closed 2016-03-26 by authmillenon