RIOT-OS/RIOT #4387: conn: avoid duplicate binding

OlegHahm, 2015-12-02:
`conn` doesn't check if an address-port tuple is already used ("bound" in POSIX terminology) by another thread. One way to do so, would be by storing pointers to all created `conn` structs and iterate over this list on creation. A potential problem here is that the "owner" of this struct might free it without telling `conn`, i.e. calling `close()`.

authmillenon, 2015-12-02:
Since most stacks (except GNRC) already provides this functionality (which lead to problems in lwIP tests with `conn_.*_sendto()` btw, because you can't use `sport` for an already established connection), I think this is more a task for the stack. For GNRC the fix for that could be way simpler: if nettype TCP or UDP (or any other connection type using ports) then only allow one port as demux_ctx in `gnrc_netreg`. This makes registration to `netreg` slightly slower (because for those types the list needs to be iterated), but it shouldn't be more than 3-jump instructions + an extra return.

OlegHahm, 2015-12-02:
But binding a port to different addresses on the same host is valid behaviour.

authmillenon, 2015-12-02:
Ah, you're right.

OlegHahm, 2015-12-02:
Nevertheless, I think it could be okay for gnrc not to support this.

authmillenon, 2015-12-02:
Especially, since acceptance based on an (addr, port) pair is not something GNRC supports (GNRC lazy-ORs those two, not ANDs them)

authmillenon, 2015-12-02:
Lazy-ORing is not the right word, but I guess you know what I mean.

authmillenon, 2015-12-02:
It checks if the packets destination is on an interface and it checks if the packets destination port has a subscriber, but it does not check if the subscriber is bound to both on the same time.

OlegHahm, 2015-12-02:
Yepp, in that case I would be fine with the proposed solution.

authmillenon, 2016-04-19:
#5091 was already moved to the next release so let's move this one too.

OlegHahm, 2016-04-19:
Did you put it into the known issue list? (I always prefer to keep issues tagged for the release until they're listed as a known issue in the release notes - if applicable.)

authmillenon, 2016-04-19:
Aren't any issues that aren't solved for the release known issues?

authmillenon, 2016-04-19:
Also, I'm still unsure if this is really a bug and not a feature ;-)

OlegHahm, 2016-04-19:
> Aren't any issues that aren't solved for the release known issues?

Yes, but you need to track them somehow.

authmillenon, 2016-04-19:
https://github.com/RIOT-OS/RIOT/issues?utf8=%E2%9C%93&q=is%3Aopen+is%3Aissue+-milestone%3A%22Release+2016.04%22+

OlegHahm, 2016-04-19:
Exactly this is why I like to keep them tagged for the release until they're noted down in the release notes.

authmillenon, 2016-04-19:
Tagged as what?

OlegHahm, 2016-04-19:
"Release 2016.04" in this case

authmillenon, 2016-04-19:
Yes, but what does it mean compared to "Known Issue" (aka issue not solved at release time)?

OlegHahm, 2016-04-19:
It's just much easier to list them.

authmillenon, 2016-04-19:
I still don't get, why [the search for "All issues, except the ones for Milestone 'Release 2016.04'"](https://github.com/RIOT-OS/RIOT/issues?utf8=%E2%9C%93&q=is%3Aopen+is%3Aissue+-milestone%3A%22Release+2016.04%22) isn't exactly the list of known issues and why we have to specially mark them (with the milestone for the release no less).

OlegHahm, 2016-04-19:
Usually it should be the other way around: all issues for the Milestone "Release 2016.04" is the list of known issues.

authmillenon, 2016-04-19:
But where go the other issues then?

OlegHahm, 2016-04-19:
If it's not tagged for the release, it's not relevant to be listed as a known issue (e.g. a feature request).

authmillenon, 2016-04-19:
I think this way some known issues get lost... You can also exclude labels in the search.

authmillenon, 2016-04-19:
Some known issues for 2015.12 are still marked with the milestone for that release (or even unmarked).

authmillenon, 2016-04-19:
But if this is your workflow, I won't criticize it more. Re-tagged it for this release.

kaspar030, 2016-04-19:
> If it's not tagged for the release, it's not relevant to be listed as a known issue (e.g. a feature request).

But open issues from one release should all get re-tagged...

cgundogan, 2016-04-19:
>But open issues from one release should all get re-tagged...

Once they were noted down in the release notes, they can be re-tagged AFAIU.

authmillenon, 2016-04-19:
They can, but some aren't (just click on the known issues in the last release and see)

cgundogan, 2016-04-19:
That's because human do err. Old issues should be verified for the current
release, retagged appropriately and added to the known issues.

authmillenon, 2016-04-19:
yeah, but as far as I can see there are also old issues never mentioned in "known issues" (need to recheck).

kaspar030, 2016-04-19:
Maybe we should go back to a "RIOT next" milestone. Everything postponed gets retagged there. Everything that *must* be fixed can be tagged with a current release milestone. No release with open issues.

... that way we don't get stale issues...

haukepetersen, 2016-04-20:
retaggt to 2016.07, I will worry about collecting it for the open issue list...

miri64, 2016-07-11:
See #5091...

kYc0o, 2016-07-25:
There's no milestone for #5091, so I'll postpone this one.

miri64, 2016-10-17:
Fixed in #5772.

closed 2016-11-04 by miri64