RIOT-OS/RIOT #4388: POSIX sockets: open socket is bound to a specific thread

OlegHahm, 2015-12-02:
With the `gnrc_conn` based implementation of POSIX sockets the socket is bound to a specific thread, e.g. reading from it will only work for the the thread that performed the binding (implicitly or explicitly). I think the most promising solution to solve this would be a thread-decoupled message queue as @kaspar030 or @authmillenon proposed.

haukepetersen, 2016-04-20:
postponed, same as #4474

kYc0o, 2016-07-25:
Sadly, postponed again.

miri64, 2016-10-17:
Fixed by #5772 and my [currently WIP posix socket port to sock](https://github.com/miri64/RIOT/tree/posix/enh/use-sock-for-sockets)

OlegHahm, 2017-01-15:
wontfix

miri64, 2017-01-15:
Shouldn't we at least wait until #5772 got merged?

OlegHahm, 2017-01-15:
No strong opinion. I saw this issue as conn related (part of the description) and hence as a wontfix.

miri64, 2017-01-24:
Closed via #6004.

closed 2017-01-24 by miri64