RIOT-OS/RIOT #4397: gnrc_sixlowpan: Header order or UDP and IPv6 swapped

DipSwitch, 2015-12-03:
One would expect that when you receive a packet the header order would be like: `data -> UDP -> IPv6`

However the header is sometimes swapped to: `data -> IPv6 -> UDP`
To search the proper header one must `LL_SEARCH_SCALAR`

I can't reproduce this behavior on my development board STM32F1xx with an AT86RF212B but on the SAM R21 Xpro this seems to happen more often.

More info about this issue can be found on: https://github.com/RIOT-OS/RIOT/pull/3911#issuecomment-157683297

authmillenon, 2015-12-07:
Correct me if I'm wrong, but wasn't that a problem related to #4189? Shouldn't this be fixed there before it got merged?

OlegHahm, 2015-12-07:
I think it's not 100% clear if it is related to #4189, but we should definitely check.

authmillenon, 2015-12-15:
Found the bug: https://github.com/RIOT-OS/RIOT/pull/4189/files#r47666253

OlegHahm, 2015-12-15:
Nice job!

DipSwitch, 2015-12-15:
Whoohoo, good going! :D

DipSwitch, 2015-12-20:
So this can be closed?

authmillenon, 2015-12-20:
No, the PR isn't merged yet.

PR #4485 (2015-12-22): gnrc_sixlowpan: Fix IPHC/NHC packet order problem
Fixes #4397 (and maybe #4462). Could not test because for some reason I can't send anything right now (even with `txtsnd`).

Here is the general idea of this fix: the `ipv6` parameter of `gnrc_sixlowpan_iphc_decode()` is replaced with the out-parameter `dec_hdr`. While `gnrc_sixlowpan_iphc_decode()` handles the dereferenced pointer the same as it used to handle `ipv6` `iphc_nhc_udp_decode()` sets it to the UDP header and appends the IPv6 header (the old value of `dec_hdr`) to the UDP header. This way the correct header order should be preserved [... cut]
closed 2015-12-22 by cgundogan