RIOT-OS/RIOT #4409: unsigned char casting in uart_getchar() implementations

Kijewski, 2015-12-05:
The signature is `int uart_getchar(FILE *)`. The result value should not be casted to an `(int)` (the automatic cast), but `(int) (unsigned char)` == `(unsigned char)`. Otherwise problems with UTF-8 or binary input may occur.

authmillenon, 2016-04-20:
As far as I can see this is an Arduino-related issue, right?

authmillenon, 2016-04-20:
```
$ git grep uart_getchar
boards/arduino-mega2560/board.c:static int uart_getchar(FILE *stream);
boards/arduino-mega2560/board.c:static FILE uart_stdin = FDEV_SETUP_STREAM(NULL, uart_getchar, _FDEV_SETUP_READ);
boards/arduino-mega2560/board.c:int uart_getchar(FILE *stream)
```

Kijewski, 2016-04-20:
Actually, I don't know anymore. The only occurrence of `uart_getchar()` is implemented correctly: [board.c#L80](https://github.com/RIOT-OS/RIOT/blob/b5497c3be43bdfecb0fb10e84bffbbd76d7a1ca1/boards/arduino-mega2560/board.c#L80).

Closed because obsolete.

closed 2016-04-20 by Kijewski