RIOT-OS/RIOT #4418: network: provide convenient functions

OlegHahm, 2015-12-06:
Currently, a lot of shell commands provide some convenient functions (mostly for networking stuff) like "printing interface information", "print the routing table", or "send a certain packet". I think it would be useful to factor out these functions to be used outside the shell. If, e.g., one wants to build an application without the shell but prints the configured addresses (like the [microcoap example](https://github.com/RIOT-OS/RIOT/tree/master/examples/microcoap_server)) or wants to send a simple link-layer packet, this could be useful. Otherwise either code from shell commands gets duplicated or shell commands are called directly without a documented and well-defined interface.

PeterKietzmann, 2015-12-07:
I would have already used this several times

BytesGalore, 2015-12-07:
@OlegHahm something like providing headers for the shell commands?

OlegHahm, 2015-12-07:
> something like providing headers for the shell commands?

Pretty much, yes. I think the Contiki folks have something like that in https://github.com/contiki-os/contiki/tree/master/apps

kYc0o, 2016-07-26:
AFAIK this issue is not completely solved, so I'll move it for the next release.

miri64, 2016-10-17:
Fixed title ;-)

kaspar030, 2016-11-18:
This issue is very broad. How about we compile a list of those functions and tackle them one-by-one?

@OlegHahm I hope you don't mind but I'll start.

OlegHahm, 2016-11-18:
I think this makes a lot of sense. Thanks!

PeterKietzmann, 2019-02-11:
> @OlegHahm I hope you don't mind but I'll start.

@kaspar030 did you?

kaspar030, 2019-02-11:
> @kaspar030 did you?

It seems to me back then I just edited the issue and added some.