RIOT-OS/RIOT #4427: vagrant: flashing ATSAMR21 fails

emmanuelsearch, 2015-12-07:
Though it worked the very first time I tried to flash/term, it stopped working after that, and I get the following error when I try to flash:

```
### Flashing Target ###
Open On-Chip Debugger 0.9.0 (2015-09-02-10:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: error writing data: (null)
Error: CMSIS-DAP command CMD_INFO failed.
```

cgundogan, 2015-12-07:
might be a long shot, but: Did you try this with an usb 3.0 port? If yes, could you use an usb 2.0 port?

emmanuelsearch, 2015-12-07:
mmh. Not so sure how to do that on my Macbook which has only USB3.0 external ports. I tried to connect via an external USB 2.0 hub (which effectively has the devices listed as connected through USB 2.0 in the system) but:
- it still does not work and give the same error
- I suspect going through this USB hub and then through the USB 3.0 port of the Macbook is not helping in other ways...

daniel-k, 2015-12-08:
@emmanuelsearch 
You can try a recent development version of OpenOCD, maybe this helps

emmanuelsearch, 2015-12-11:
@daniel-k tried with 0.10.0-dev version of OpenOCD, still does not work and throws the same type of error (though it seems to use a different speed 400kHz instead of 500kHz with 0.9, no idea why)

```
### Flashing Target ###
Open On-Chip Debugger 0.10.0-dev-00146-g332023f (2015-12-11-13:59)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Error: error writing data: (null)
Error: CMSIS-DAP command CMD_INFO failed.
```

kaspar030, 2015-12-11:
@emmanuelsearch Could you try if https://github.com/ataradov/edbg is able to flash?

emmanuelsearch, 2015-12-11:
Thanks for the tip. Here's what I get: first time goes to timeout, and then it does not find the board anymore.

```
vagrant@vagrant:~/RIOT-Tutorial/application/bin/samr21-xpro$ edbg -bpv -t atmel_cm0p -f ./ieee_wf-iot_2015.elf
debugger write(): Connection timed out
vagrant@vagrant:~/RIOT-Tutorial/application/bin/samr21-xpro$ edbg -bpv -t atmel_cm0p -f ./ieee_wf-iot_2015.elf
Error: no debuggers found
```

emmanuelsearch, 2015-12-11:
(for the record I'm in the mean time I'm able to flash directly from OS X  woohoo ;)

emmanuelsearch, 2015-12-11:
(but it would still be even better that this VM would work for everyone, of course)

bapclenet, 2016-02-23:
I have encountered the same problem:
```
### Flashing Target ###
Open On-Chip Debugger 0.9.0-dev-00357-g09ca5af-dirty (2015-11-25-17:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.sourceforge.net/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: CMSIS-DAP command CMD_INFO failed.
in procedure 'init'
Done flashing
```

cgundogan, 2016-02-23:
@kYc0o may have found a solution for this?

kYc0o, 2016-02-23:
Maybe it works just by chance but try uninstalling:
```
sudo apt-get remove --purge libusb-dev libusb-1.0-0-dev libhidapi-dev
```
And also be sure that USB 3.0 is not selected in the VM (you can use USB 1 or 2)

bapclenet, 2016-02-24:
I shut down my VM and my computer, restart everything and it works (without uninstalling what @kYc0o  said)

emmanuelsearch, 2016-03-02:
Actually, it seems to work for me too now, even if I have not uninstalled  what @kYc0o was pointing at.
Shall I close? Or do some people still have problems?

bapclenet, 2016-03-02:
There is still a problem but I don't think it is RIOT dependant so I think we should close it

cgundogan, 2016-03-02:
that's very interesting .. I wonder what changed. Was there an OS X update recently? Or did your OpenOCD version change?

@emmanuelsearch in your initial report you had the OpenOCD version
```
Open On-Chip Debugger 0.9.0 (2015-09-02-10:42)
```

Did that change?

cgundogan, 2016-03-02:
@thomaseichinger can you also verify that flashing from within vagrant works flawlessly on your osx machine?

cgundogan, 2016-03-02:
@bapclenet do you encounter an error while writing to the memory?

kYc0o, 2016-03-02:
Sometimes when I have issues while flashing I try to unload usb cdc or ftdi drivers, using kextunload

BytesGalore, 2016-03-03:
It sounds like the OpenOCD has not been built with the proper flags set.
Could you try to build a custom OpenOCD  as described in [1]?

[1] http://watr.li/samr21-dev-setup-ubuntu.html#building-openocd

bapclenet, 2016-03-03:
@cgundogan, I had this error once, I restart my VM and Computer and I can't reproduce the error, sorry

cgundogan, 2016-03-03:
@jia200x can you also verify that flashin the samr21-xpro from within vagrant on your osx machine works?

jia200x, 2016-03-04:
@cgundogan for me it's not working with Vagrant. I get the same "CMSIS-DAP command CMD_INFO failed.". Also tried with USB 2.0.

cgundogan, 2016-03-23:
I close this issue as the majority of OSX people around here have no trouble anymore. It seems to be a problem on the host system and not with the vagrant machine itself. I will tag this issue as memo.

closed 2016-03-23 by cgundogan

WulfricLee, 2019-02-16:
I've encountered this problem before, it's very weird in my case, I plug a USB wire into my CMSIS-DAP debugger and a USB HUB which connected to my computer, at first I plugged it in the 2nd slot of my USB HUB, and I ran OpenOCD, I was told
```
Error: CMSIS-DAP command CMD_INFO failed.
```
then I tried the first slot, it just worked! And I try the second slot again, the error prompted to me as before, I tried each slot in my HUB, the result is different everytime, sometimes even the first slot prompted the error to me also, I was like... What the heck???
[image]
And suddenly I realized the USB wire I used to connect debugger and HUB is like... more than 1 meter long, so maybe it is a bug caused by electrical characteristics of my USB wire, I replaced the wire with another one which is about 20cm long, and the result is, **whichever** slot I plugged in, it just worked.
So, here is my advisement:
### If your USB wire is more than 0.5 meter, replaced the wire with a shorter one.