RIOT-OS/RIOT #4434: OpenMote Makefile should support flashing via cc2538-bsl tool during "make flash"

aeneby, 2015-12-08:
The cc2538dk board supports flashing via the dist/tools/cc2538-bsl/cc2538-bsl.py script directly from the Makefile. In fact, it's the default flashing method for this platform - you need to explicitly specify if you want to flash via JTAG instead using "PROGRAMMER=jlink".

This capability (and default behaviour) should be carried across to the OpenMote Makefile too.

thomaseichinger, 2015-12-09:
@aeneby Thanks for bringing this up. For consistency I certainly prefer your proposal, last time I checked though, the OpenMote on my desk was not flashable with this script although I used the "pin-magic". Do you know if there are different versions of the platform?

aeneby, 2015-12-10:
@thomaseichinger I have successfully flashed RIOT to my OpenMote using the cc2538-bsl script manually, via the OpenBase board. AFAIK there is only one version of the OpenMote platform right now. I will try to provide a patch for you.

thomaseichinger, 2016-04-03:
Fixed by #4460.

closed 2016-04-03 by thomaseichinger