RIOT-OS/RIOT #4439: stm32f4: pwm_init returns wrong value

A-Paul, 2015-12-08:
In *cpu/stm32f4/periph/pwm.c* `pwm_init` is returning the requested input value.
It should return a frequency that is calculated based on actual clock and prescaler values.

haukepetersen, 2015-12-08:
not sure, but isn't the implementation actually correct? As the STM32F4 has very fine means for defining the actual clock, I think the configured value is the same (or at leas very very close) to the requested value.

haukepetersen, 2015-12-08:
but then again - I implemented it and thought this then also, I might be wrong here... Do you maybe have an example where the returned value differs from the value set?

A-Paul, 2015-12-08:
> defining the actual clock, I think the configured value is the same (or at leas very very close) to the requested value.

This is integer division. They can't be the same. Except in special cases, when *res * freq* is a divisor of the clock frequency. The differences are growing with the value of that product.

haukepetersen, 2015-12-09:
Ok, forgot what I wrote above, I had some twists in my thoughts, you are completely right!

haukepetersen, 2015-12-09:
how about this fix? #4454

PR #4454 (2015-12-10): cpu/stm32f4discovery: fixed freq return in pwm_init
fixes #4439
closed 2015-12-10 by A-Paul