RIOT-OS/RIOT #4451: pkg: tlsf: not building with -pedantic

OlegHahm, 2015-12-09:
gcc with pedantic doesn't like the @Kijewski's patch:
`/home/oleg/git/RIOT/examples/ccn-lite-relay/bin/native/tlsf-src/tlsf.c:161:6: error: ISO C forbids specifying range of elements to initialize [-Werror=pedantic]
   [0 ... FL_INDEX_COUNT-1] = {`

@Kijewski, would you object reverting this part of the patch?

The corresponding code looks like:
```C
/* The TLSF control structure. */
typedef struct control_t
{
    /* Empty lists point at this block to indicate they are free. */
    block_header_t block_null;

    /* Bitmaps for free lists. */
    unsigned int fl_bitmap;
    unsigned int sl_bitmap[FL_INDEX_COUNT];

    /* Head of free lists. */
    block_header_t* blocks[FL_INDEX_COUNT][SL_INDEX_COUNT];
} control_t;

static control_t control __attribute__ ((aligned(ALIGN_SIZE))) = {
    .block_null = {
        .next_free = &control.block_null,
        .prev_free = &control.block_null,
    },
    .fl_bitmap = 0,
    .sl_bitmap = {0},
    .blocks = {
        [0 ... FL_INDEX_COUNT-1] = {
            [0 ... SL_INDEX_COUNT-1] = &control.block_null
        }
    }
};
```

Kijewski, 2015-12-09:
If I remember correctly I did that so I could throw away an initialization function because I found no proper point when to call it. Newlibs print functions (incl. puts if I am not mistaken) already use malloc, so the initialization has to be done before the first line is printed.

OlegHahm, 2015-12-16:
Oops, I just realized that this was a duplicate of #4265.

PR #4455 (2015-12-16): pkg tlsf: make it compile with -pedantic
Fixes #4451 by re-introducing the init function and updates using functions accordingly.
closed 2015-12-16 by OlegHahm