RIOT-OS/RIOT #4462: IPHC/NHC broken

cgundogan, 2015-12-10:
Following setup:
Linux <-> `gnrc_border_router` example (N1) <-> `microcoap_server` example (N2)

I am able to ping N2 from linux, but I am not able to make CoAP requests.
If I disable iphc with `ifconfig x -iphc` on N1 and N2, then I can do CoAP requests (with Copper) from Linux <-> N2.

I have no time to investigate further right now. Can anyone confirm this behavior? @kaspar030 didn't you state something similar on IRC yesterday?

kaspar030, 2015-12-10:
> @kaspar030 didn't you state something similar on IRC yesterday?

Yep, exactly the same. Wireshark shows a garbled CoAP reply arriving in Linux. Disabling iphc makes it work.

OlegHahm, 2015-12-10:
Can someone check with TFTP or simple netcat? Should be affected, too. But I saw it working while reviewing the NHC PR.

authmillenon, 2015-12-11:
Was this problem apparent before https://github.com/RIOT-OS/RIOT/pull/4189 got merged?

OlegHahm, 2015-12-11:
Before this problem, NHC wouldn't work at all. Since IPHC (including NHC) is enabled per default on Linux, no UDP connection between RIOT and Linux did work.

kaspar030, 2015-12-11:
Isn't IPHC sixlowpan specific?

OlegHahm, 2015-12-11:
Yes.

OlegHahm, 2015-12-11:
Ah, ah, I see. I meant `no UDP connection over 802.15.4 link layer`.

kaspar030, 2015-12-11:
Ok, I got that right. In the setup above, linux <-> 6lr is using IPv6 over ethernet. Seems like 6lr doesn't unpack IPHC stuff correctly when handing it to plain IPv6.

authmillenon, 2015-12-11:
Since the issue is only with UDP traffic and this behavior wasn't appearent with our 2015.09 tests, I strongly suspect that NHC is the problem.

jfischer-phytec-iot, 2015-12-11:
@cgundogan Can you please comment  ```USEMODULE += gnrc_sixlowpan_iphc_nhc``` out and test again?

cgundogan, 2015-12-12:
I can confirm that Linux (netcat) -> N2 (posix udp server) works with iphc. N2 -> Linux doesn't work. This is however a completely different problem (`_implicit_bind` fails because `gnrc_conn6_set_local_addr` returns false).

Nevertheless, CoAP ping/discovery is broken with iphc enabled and works when iphc is disabled. "Normal" udp traffic (induced with netcat) does seem to work with and without iphc enabled.

cgundogan, 2015-12-12:
UDP traffic (posix example - single hop) between two RIOT nodes seems to be broken when iphc is enabled. Disabling iphc on the sending node fixes the problem.

cgundogan, 2015-12-12:
@jfischer-phytec-iot  with `DISABLE_MODULE += gnrc_sixlowpan_iphc_nhc` CoAP works without disabling iphc manually via the ifconfig command

authmillenon, 2015-12-13:
Maybe related to #4397?

authmillenon, 2015-12-18:
Found another issue, that might cause this: Currently IPv6 assumes for forwarding, that nothing is between the IPv6 header and the payload for the pktsnip reversal (https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L824). This is not the case with NHC, since UDP will be between the two. Did not go into it in-depth and did not have a fix yet.

cgundogan, 2015-12-22:
not fixed

OlegHahm, 2015-12-22:
Good catch, thanks

jfischer-phytec-iot, 2015-12-22:
Have wireshark running while testing https://github.com/RIOT-OS/RIOT/pull/4485#issuecomment-166650528, as I see it, the Border Router does not perform NHC decoding.

**riot-node -> riot-6lo**
[image]

**riot-6lo -> linux**
[image]

authmillenon, 2015-12-22:
is it a linux br or a riot br?

jfischer-phytec-iot, 2015-12-22:
riot br (riot-6lo - (over tun) -> linux host)

jfischer-phytec-iot, 2015-12-22:
@authmillenon Oh no, opposite, riot-6lo makes nhc encoding for slip-iface, or?

authmillenon, 2015-12-22:
That would be very weird, because the slip interface shouldn't even go near 6LoWPAN (and why is IPHC not done there too).

OlegHahm, 2015-12-22:
Maybe the slip interface doesn't remove the NHC snip while forwarding or so?

jfischer-phytec-iot, 2015-12-22:
The second dump looks exact like encoded UDP, src + dst inline, length elided, payload ok. Next Header is wrong ...

authmillenon, 2015-12-22:
I'm sorry, got to go… wasn't able to set up my border router (samr21-xpro with https://github.com/RIOT-OS/RIOT/pull/4212) properly and don't have a UART adapter right now :-/. I suspect the NHC decoding on forwarding not working, other explanation would be very weird. Further investigation into this by others would be much appreciated :-)

jfischer-phytec-iot, 2015-12-22:
There is no UDP module in the examples/gnrc_border_router, no UDP no NHC decoding :smile:

cgundogan, 2015-12-22:
even after compiling `gnrc_udp` in it still crashes at the 6lr node.. don't know why

jfischer-phytec-iot, 2015-12-22:
6lo do not forward the udp packet to the node...

authmillenon, 2015-12-22:
> There is no UDP module in the examples/gnrc_border_router, no UDP no NHC decoding :smile:

Why? This is a perfectly good use-case, why NHC without UDP would make sense. If we remove this dependency this should work, right?

cgundogan, 2015-12-22:
my 6lr node crashes, because it tries to send the decoded message to the ipv6 thread (`gnrc_netapi_dispatch_receive()`), but it crashes because the pkt gets traversed in the following order (nettype):
0 (undef) => 4 (udp) => 2 (ipv6) => 65 (???) => 42 (???) => crash

jfischer-phytec-iot, 2015-12-22:
@cgundogan based on #4485?

>Why? This is a perfectly good use-case, why NHC without UDP would make sense. If we remove this dependency this should work, right?

no, the node (microcoap) sends encoded udp, the 6lo needs nhc to decode it.

cgundogan, 2015-12-22:
btw I am doing my testing with #4485 merged

jfischer-phytec-iot, 2015-12-22:
crash == hard fault ?

cgundogan, 2015-12-22:
yes and pc showing to nirvana

cgundogan, 2015-12-22:
```
2015-12-22 18:33:57,995 - INFO # Context before hardfault:
2015-12-22 18:33:57,996 - INFO #    r0: 0x0000000a
2015-12-22 18:33:57,997 - INFO #    r1: 0x2000004c
2015-12-22 18:33:57,998 - INFO #    r2: 0x00000000
2015-12-22 18:33:57,999 - INFO #    r3: 0x20006355
2015-12-22 18:33:57,999 - INFO #   r12: 0x00000008
2015-12-22 18:33:58,000 - INFO #    lr: 0x0000079b
2015-12-22 18:33:58,001 - INFO #    pc: 0x00005fb6
2015-12-22 18:33:58,002 - INFO #   psr: 0x21000000
2015-12-22 18:33:58,002 - INFO #
2015-12-22 18:33:58,003 - INFO # Misc
2015-12-22 18:33:58,004 - INFO # EXC_RET: 0xfffffffd
2015-12-22 18:33:58,006 - INFO # Attempting to reconstruct state for debugging...
2015-12-22 18:33:58,007 - INFO # In GDB:
2015-12-22 18:33:58,009 - INFO #   set $pc=0x5fb6
2015-12-22 18:33:58,010 - INFO #   frame 0
2015-12-22 18:33:58,011 - INFO #   bt
```

jfischer-phytec-iot, 2015-12-22:
6lo over ethernet ? (works for me on samr21)

cgundogan, 2015-12-22:
my setup is: linux (tunslip) <=> gnrc_boarder_example with gnrc_udp module <=> coap example

cgundogan, 2015-12-22:
also testing with samr21

cgundogan, 2015-12-22:
and with #4485 merged

authmillenon, 2015-12-22:
> no, the node (microcoap) sends encoded udp, the 6lo needs nhc to decode it.

Yes, but the border router doesn't need the UDP module, because it only forwards. It does need NHC however to decode properly for SLIP/Ethernet traffic. So the dependency between NHC module and the UDP module must not be.

jfischer-phytec-iot, 2015-12-22:
>Yes, but the border router doesn't need the UDP module, because it only forwards. It does need NHC however to decode properly for SLIP/Ethernet traffic. So the dependency between NHC module and the UDP module must not be. 

agree, but it exist
```
#ifdef MODULE_GNRC_UDP
            case NHC_UDP_ID:
                printf("6lo iphc: do decode nhc\n");
                payload_offset = iphc_nhc_udp_decode(pkt, dec_hdr, payload_offset);
                break;
#endif
```

jfischer-phytec-iot, 2015-12-22:
@cgundogan have the same test setup

```
b398941 gnrc_udp: fix false assumed packet order
c2d9342 gnrc_sixlowpan: Fix IPHC/NHC packet order problem
b7a7578 Merge pull request #4507 from authmillenon/gnrc_ipv6/fix/no-order-assumptions
83649dd Merge pull request #4531 from Yonezawa-T2/gnrc_avoid_infinite_loop
34974ab Merge pull request #4534 from Yonezawa-T2/gnrc_off_by_1_overrun
5b4ef43 Merge pull request #4532 from Yonezawa-T2/gnrc_safe_no_routers
08ba1f8 gnrc_ipv6_netif: fixed buffer overrun
fe0e58d gnrc_sixlowpan_nd: fixed crash when there are no routers
9108d27 sixlowpan: fixed infinite loop
2d8f6f8 Merge pull request #4500 from haukepetersen/fix_doc_samd21
1ac9a0d Merge pull request #4480 from authmillenon/doc/enh/creating-modules
b5f5f83 doc: add page about creating modules
```

cgundogan, 2015-12-22:
I just did a clean setup and removed all my custom debug statements and merged #4485 again: the same hard fault

cgundogan, 2015-12-22:
BTW: I use the copper firefox plugin to do a CoAP ping. When issueing that CoAP ping, the 6lr crashes

jfischer-phytec-iot, 2015-12-22:
ok, I use libcoap and ```coap-client coap://[affe::d173:4e02:71af:600a]/riot/board```

cgundogan, 2015-12-22:
current status: with `gnrc_udp` compiled into the boarder router, the 6lr node crashes for me when I use any kind of CoAP traffic over the 6lbr. I use samr21-xpros for both nodes. @jfischer-phytec-iot uses a samr21-xpro for the 6lbr and a phynode for the 6lr and his 6lr does not crash when using CoAP traffic.

cgundogan, 2015-12-22:
(on master + #4485)

jfischer-phytec-iot, 2015-12-23:
I have found an other bug in iphc_nhc_udp_encode...

cgundogan, 2015-12-23:
>I have found an other bug in iphc_nhc_udp_encode...

enlighten us please

jfischer-phytec-iot, 2015-12-23:
wip

authmillenon, 2015-12-23:
But #4485 fixed @kaspar030's and @cgundogan's issue?

authmillenon, 2015-12-23:
@jfischer-phytec-iot can you at least sketch out a bug description so we can assess the severity for the release?

cgundogan, 2015-12-23:
> But #4485 fixed @kaspar030's and @cgundogan's issue?

Nope, it didn't fix @kaspar030's and my issue.

jfischer-phytec-iot, 2015-12-23:
I need 20-40 Minutes for a PR, i hope ...

There is also an possible overflow in gnrc_pktbuf_realloc_data, is it known?

authmillenon, 2015-12-23:
Afaik no.

jfischer-phytec-iot, 2015-12-23:
@cgundogan can you please test #4544 ?

gebart, 2016-03-02:
@cgundogan did you test #4544 ?

authmillenon, 2016-03-02:
Maybe #4935 will also help ;-).

authmillenon, 2016-04-19:
This issue is fixed.

authmillenon, 2016-04-19:
(with merging of #5232)

PR #4507 (2015-12-22): gnrc_ipv6: make no pre-assumption about any marked headers
One more PR that aims to fix #4462 (might need #4485 to fully fix the issue). With NHC activated the highest layer payload isn't directly adjacent to IPv6 anymore nor is the IPv6 header necessarily the first or second snip in a received `pkt` as it was before, since UDP (and later probably some extension headers or GHC supported headers) will be inserted between them. These assumptions are completely removed by this PR and the IPv6 header is searched as in any other layer + the pktsnip list is completely reversed for forwarding.
closed 2016-04-19 by authmillenon