RIOT-OS/RIOT #4471: boards/cpu needs refactoring to include mcu layer

punchcard60, 2015-12-13:
This is a recommendation/RFC to refactor and cleanup how hardware is defined. It is my understanding of how things currently work along with my thoughts on how it should work. I believe that implementing this would make it much easier to port new hardware and make it much easier for applications to be hardware agnostic.

The boards and cpu directories need to be refactored to include an mcu layer. Boards should reference mcu part numbers to include physical packaging since package type often changes functionality slightly and can vary between versions of the board. MCUs should in turn reference cpu types.

Boards should only include peripherals that are external to the MCU. Drivers for these devices can be found in $(RIOTBASE)/devices/saul. Peripherals that are internal to the MCU should implement the drivers in $(RIOTBASE)/devices/periph.

punchcard60, 2015-12-13:
I would like to work on this as I believe that it may benefit the Active Power Management effort.

punchcard60, 2015-12-14:
There are currently only 9 CPUs supported by RIOT: arm7, avr8, cortex-m0, cortex-m0+, cortex-m3, cortex-m4, cortex-m7 (not used?), cpuxv2 (msp430), and x86.

punchcard60, 2015-12-14:
Support for eINT, dINT, and UART0 should probably be removed as part of this effort.

kaspar030, 2015-12-14:
> The boards and cpu directories need to be refactored to include an mcu layer.

We do this mostly, e.g.,  

```boards/nucleo-f91``` depends on ```cpu/stm32f0``` and sets ```CPU_MODEL=stm32f091rc```.
```cpu/stm32f0``` includes ```cpu/cortexm_common``` and sets ```CPU_ARCH=cortex-m0```.  

While some refactoring might make this more clear, IMHO porting to new hardware is already rather easy.

> applications to be hardware agnostic.

Do you have examples where applications are not hardware agnostic?

punchcard60, 2015-12-14:
Most applications that want to use features that aren't defined will just implement them in the application instead of going through the trouble of opening an issue and waiting for implementation or learning how to become a RIOT developer so they can submit a PR. So I don't have specific examples because I don't have access to any other applications, but I would bet a large sum of money that they are out there.

I believe that new hardware implementation could be even easier than it is. Right now there is a fair amount of "not the best way" and duplicate implementation that could be cleaned up. Things like having MCU internal definitions inside board.h (defining which CPU is inside the MCU). I believe that a new structure would help with the cleanup and make it easier to implement some of the missing and future features.

## Implementation Proposal 1
Leave the boards directory and the cpu directory where they are. Modify the makefiles so that defining NEW_HARDWARE will cause the "new" board and cpu files to be used. Add a deprecation message to the "old" board files. This way there will be minimal disruption to applications that have directly manipulated hardware through the defines in these files.

Create the following directory/Makefile.include structure for the new cleaned-up files:

    hardware 
    +-- include    (contains the generic peripheral headers from devices/periph)
    +-- boards
    |    +-- Makefile.include
    |    +-- abc_board
    |    |   +-- Makefile.include
    |    |   +-- board.h  
    |    ...
    |    +-- xyz_board
    |       +-- Makefile.include
    |       +-- board.h
    +-- mpu
    |   +-- Makefile.full_part_number.include
    |   ...
    |   +-- Makefile.full_part_number.include
    |   +-- family_part_number		(directory contains resources (.h, .c) for the part family)
    +-- cpu
    |    +-- cortex_m0
    |    …
    |    +-- cortex_r
    |    +-- avr8
    |    +-- avr32
    |     …
    |

kaspar030, 2015-12-14:
Well, if in your graph you move all cpu/<cpu> stuff into mcu/<cpu>_common, you pretty much describe our file layout.

We could factor out some of the cpu/ Makefiles' definitions into part specific files, but that would probably create more files than it would save lines. Currently, boards select MCU part numbers together with the MCU family (```CPU=stm32f3, CPU_MODEL=sm32f334r8```).

IMHO we do have some MCU specific defines in board.h as that bundles the hardware configuration.

> Most applications that want to use features that aren't defined

Are you referring to missing definitions for e.g., an MCU's timer on top what's in board.h?

punchcard60, 2015-12-14:
I know you guys have put a LOT of work into RIOT and it shows. But, the easier and faster it is for the users to learn and begin to modify/port/use RIOT, the faster and farther it will propagate. 

Instead of burying it inside makefiles, why not make the part number information visible for anybody that wants to port RIOT to a new board? The proposed implementation would also make it cleaner and easier to share code. For example STM32F407 and STM32F415 are around 90% the same so why have two 600K .h files to define them? All of the STM32F405xx, ...407xx, ...415xx, and ...417xx MCUs could use the same register definitions. Why not have one stm32F4xxxx.h and a handful of very small makefiles that call out the differences for each particular part?

"Missing" stuff refers to Event Bridges and similar features.

??? Side question: How would an application set up a DMA transfer from say an I2C port to an SPI port?

I disagree with putting MCU internal definitions in board.h. They should be at the MCU level so they can be used by any code that uses that MCU without having to duplicate code or re-arrange the code for multiple boards.

kaspar030, 2015-12-14:
> For example STM32F407 and STM32F415 are around 90% the same so why have two 600K .h files to define them? All of the STM32F405xx, ...407xx, ...415xx, and ...417xx MCUs could use the same register [...]

Well, in this case, those files are vendor-supplied. The vendor probably generates these headers from some saner (non-C) register description. Doing that by hand is a lot of work... So if in these cases (a bunch of possibly half redundant register definitions) we usually go with the vendor headers unless that leads to actual code duplication. And even if we'd kill redundancy in those files, what is there to gain? Actual code will have to make the same (or more) distinctions as before.

> Instead of burying it inside makefiles, why not make the part number information visible for anybody that wants to port RIOT to a new board?

Well, our current build system does selective compiles by Makefile variables. Not everything can easily be expressed in C. I don't think that is "burying in makefiles". The board makefile clearly sets e.g., "CPU=stm32f0", "CPU_MODEL=stm32f091rc", which implicitly includes "cpu/stm32f0" stuff. If you know where to look (e.g., stmf4discovery uses "stm32f407vg"), you might miss a "stm32f407vg.c", but actually all that code is nicely isolated in "cpu/stm32f4".

I totally agree that there's a lot of cleanup and consolidation that can be done, and a lot of work has gone in that direction this year.

> I disagree with putting MCU internal definitions in board.h.

Probably board.h for some boards contains stuff that can and should be moved to the cpu/mcu level. If you find something, open an issue, or better, PR a fix.

> "Missing" stuff refers to Event Bridges and similar features.
> ??? Side question: How would an application set up a DMA transfer from say an I2C port to an SPI port?

These are concepts that are hard to put into a generic API, or that we didn't come to think about yet. Not sure those are related to the refactoring issue.

kaspar030, 2015-12-14:
> Support for eINT, dINT, and UART0 should probably be removed as part of this effort.

Quick "git grep" finds only one use of dINT() in the (slightly unmaintained) x86 port. "uart0" has been dropped. Any traces can probably just be removed.

DipSwitch, 2015-12-14:
The STM32F4 headers files can be merged indeed :) Not a bad idea.

kaspar030, 2016-11-18:
What do we do with this? Do we feel there's a need to refactor cpu/?

closed 2019-09-10 by stale[bot]