RIOT-OS/RIOT #4474: conn: Wishlist for Redo

authmillenon, 2015-12-14:
After a few month of working with `conn` we found some short-comings to it that need to be fixed in later iterations. This issue tries to collect these.

- connection-less support for connection-less protocols (goal: easy integration of simple lightweight stacks)
- connection-bound support for connection-less protocols if stack requires connections (goal: easy integration of full-featured stacks (e.g. lwIP))
- thread-shared connections (see https://github.com/RIOT-OS/RIOT/issues/4388)

... Feel free to add to this

DipSwitch, 2015-12-20:
Non-blocking read/write would be nice aswell :) although this could also be solved with select.

I think however that epoll would be simpler to implement than select, and could also be faster. Aswell as in the kernel / stack implementation as in application implementation.

authmillenon, 2015-12-20:
`epoll` sounds indeed nice (and could be implemented as well), but the problem is that most libraries still use `select()`. So purely from a porting perspective we need to implement it at some point anyway.

authmillenon, 2016-02-09:
The discussion regarding epoll was also held in #4222.

kaspar030, 2016-02-09:
- remove redundant address family / address length parameters (former implies the latter)
- UDP needs a way to specify source / target interface for link-local and multicast

authmillenon, 2016-02-09:
> * remove redundant address family / address length parameters (former implies the latter)

Only valid for IP-only protocols AFAWK, but true.

> * UDP needs a way to specify source / target interface for link-local and multicast

That might be do-able with options for a conn object (setsockopts style). Another feature we might need.

haukepetersen, 2016-04-20:
postponed.

kYc0o, 2016-07-25:
Postponed again...

miri64, 2016-10-17:
Realized in #5758.

closed 2016-10-17 by miri64