RIOT-OS/RIOT #4488: stm32f4: newlib hangs after calling thread_create

haukepetersen, 2015-12-16:
When calling `puts/printf` after `thread_create()`, the CPU hangs. This seems to be connected to the mutex used in the uart driver for syncing the DMA transfer in `uart_write()`. When disabling DMA for a quick test, everything works fine. So can it be, that the mutex is somehow effected when a new thread is created?

haukepetersen, 2015-12-16:
The problem is indeed related to f4's UART driver, it seems that the DMA enabled driver is not thread safe anymore...

PeterKietzmann, 2015-12-16:
Sorry, I won't find the time to check this until friday :(

DipSwitch, 2015-12-16:
I see that the atomic actions in mutex are different from what I always use. I have seen the volatile 
At work I use these: https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html and/or https://gcc.gnu.org/onlinedocs/gcc-4.4.5/gcc/Atomic-Builtins.html for ARM.

But however, I'm porting an STM32F2xx now which I base of the STM32F4xx, the board and CPU port are ready so I could take a look tomorrow evening. It uses the DMA controller to since it's the same as the F4 :)

haukepetersen, 2015-12-17:
Found the reason. The missbehavior is caused by the newlib not being threadsafe...

DipSwitch, 2015-12-17:
For newlib to be made threadsafe you need to add:

		/* Allocate a Newlib reent structure that is specific to this task.
		Note Newlib support has been included by popular demand, but is not
		used by the FreeRTOS maintainers themselves.  FreeRTOS is not
		responsible for resulting newlib operation.  User must be familiar with
		newlib and must provide system-wide implementations of the necessary
		stubs. Be warned that (at the time of writing) the current newlib design
		implements a system-wide malloc() that must be provided with locks. */
		struct 	_reent xNewLib_reent;

And then on every task swap you need to update the impure pointer.

		#if ( configUSE_NEWLIB_REENTRANT == 1 )
		{
			/* Switch Newlib's _impure_ptr variable to point to the _reent
			structure specific to this task. */
			_impure_ptr = &( pxCurrentTCB->xNewLib_reent );
		}

To every struct and implement the following functions:

	void __env_lock ( struct _reent *_r )
	{
	#if OS_THREAD_SAFE_NEWLIB
		if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED)
			return;

		vTaskSuspendAll();
	#endif /* OS_THREAD_SAFE_NEWLIB */
		return;
	}

	/* __env_unlock needs to provide recursive mutex unlocking */

	void __env_unlock ( struct _reent *_r )
	{
	#if OS_THREAD_SAFE_NEWLIB
		if (!xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED)
			return;
		  
		xTaskResumeAll();
	#endif /* OS_THREAD_SAFE_NEWLIB */
	}

	/* __malloc_lock needs to provide recursive mutex locking */

	void __malloc_lock ( struct _reent *_r )
	{
	#if OS_THREAD_SAFE_NEWLIB
		if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED)
			return;

		vTaskSuspendAll();
	#endif /* OS_THREAD_SAFE_NEWLIB */
		return;
	}

	/* __malloc_unlock needs to provide recursive mutex unlocking */

	void __malloc_unlock ( struct _reent *_r )
	{
	#if OS_THREAD_SAFE_NEWLIB
		if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED)
			return;
		  
		xTaskResumeAll();
	#endif /* OS_THREAD_SAFE_NEWLIB */
	}

This I copied from a port I made for FreeRTOS a while ago :)

haukepetersen, 2015-12-17:
Interesting, I will try to look into this, but probably not before January... I renamed the issue to reflect this.

PeterKietzmann, 2015-12-17:
(I changed the release tag to 2016.03)

PeterKietzmann, 2016-07-11:
Is anyone willing to take over the assignment of this PR? I think I won't make it any time soon. Probably @DipSwitch?
Postponing to release 2016.10

OlegHahm, 2017-01-15:
Postponing again?

RomanLudwig, 2018-08-30:
Thanks @DipSwitch for the sample code. Unfortunately the if condition in __env_lock() contains a surplus ! which causes the task scheduler to remain disabled. The if condition should be the same as in the other 3 functions.

jenswet, 2020-12-06:
I have some strange problems using RIOT with `printf`, `std::string` and other C/C++ library functions on a STM32F7. When mixing UART interaction and these functions (e.g. during argument parsing) the CPU hangs. I didn't find out yet what exactly causes the problem but it's very annoying and I am not sure how to create a stable program with the current problems. After trying a lot of things and doing some research I've finally found this issue. This seems very related.

If I get this right this should be a major issue that might cause severe problems on any newlib calls with malloc usage. Reading through #4529 and #8619 it is probably just luck that there were not many problems yet. How is the progress here? Unfortunately all PRs are quite old and difficult to try with the current master. I am not experienced enough with RIOT / OS development to understand all the thread background.

benpicco, 2021-01-10:
fixed by #15606

benpicco, 2021-01-10:
Wait, neither `printf` nor `thread_create` should call `malloc` - is this a different bug?
Please confirm if this is still an issue.

jenswet, 2021-01-10:
> Wait, neither `printf` nor `thread_create` should call `malloc` - is this a different bug?
> Please confirm if this is still an issue.

Kind of. But not in RIOT. After lot's of debugging I found out that many of the freezes I experienced were not RIOT related. The integrated USB-to-TTL STLink converter of the Nucleo I use has some bug with macOS. It can happen that the UART PC -> Nucleo direction freezes until re-plugging the USB. Sending is impossible then. 

I don't understand why this happens - but it happened much more often when using `printf` or `std` functions using malloc / creating temporary objects on the heap. At first I thought the freeze or crash was in the OS and due to malloc. This bug is also very strange because I'v never experienced it with other devices before and the serial drivers are very stable with all other devices I use no matter if FTDI, CP1202 or others.

Regarding this issue I think this bug is fixed for malloc at least. I didn't have any issues so far anymore, but didn't use it much. I focussed on using a STL implementation not relying on malloc. https://www.etlcpp.com is very interesting.

But I am not sure whether the whole newlib is alread-safe or there are other functions still missing from #4529 / #8619.

DeJusten, 2025-03-03:
> Wait, neither printf nor thread_create should call malloc - is this a different bug?
> Please confirm if this is still an issue.

printf() itself does not call malloc in the newlib-nano version.
printf(), however, uses puts() for output. In the newlib variant, puts() calls cantwrite(), which checks whether an output buffer has been created for stdout. If not, canwrite() reserves BUFSIZ=1024Byte of memory using malloc()!

This can be 'prevented' using setvbuf(), which should be called before the first call to puts()/putchar().

```
//__attribute__(constructor) ensures that premain_init() is called immediately after __sinit()
//(both within __libc_init_array())
void __attribute__((constructor)) premain_init(void)
{
#if 0
//No linebuffering, call stdio_write() immediately
//-> Slow, because __sflush_r()/_write()/stdio_write() is called with each character
setvbuf(stdout,NULL,_IONBF,0);
#else
static char linebuf[16];
//LineBuffering into global variable (good compromise)
setvbuf(stdout,linebuf,_IOLBF,sizeof(linebuf)); 
#endif
}
```