RIOT-OS/RIOT #4490: pkg: tlsf: initialize memory pool early

OlegHahm, 2015-12-16:
TLSF and should be initialized with a memory pool for its heap early on. Preferably this could be done inside the CPU startup. This would allow using TLSF's malloc implementation for basically every other module (including BSP or libc stuff) and choosing an appropriate size for the heap (i.e. RAM minus stack usage).

authmillenon, 2015-12-16:
That's also interesting in regards to lwIP. However, they bring their own version of pool-based memory allocation. We need to look into this, especially if we want to use e.g. CCN-lite together with lwIP.

OlegHahm, 2015-12-16:
I think in the long-run we should do something similar to PRNG for malloc and offer various alternative dynamic memory allocations. For CCN-Lite I was even thinking about reusing (misusing) pktbuf.

https://github.com/emeryberger/Malloc-Implementations may be a good starting point.

haukepetersen, 2016-04-20:
wont be done for this release -> postponed

kYc0o, 2016-07-26:
And postponed again...

kaspar030, 2016-09-01:
+1

Maybe using the linker variables to find out about unused memory areas?

OlegHahm, 2016-09-01:
Yes, if I remember correctly that was also an approach that @haukepetersen and me discussed some time ago.

jcarrano, 2018-07-18:
Please see https://github.com/jcarrano/RIOT/blob/tlsf-init-experiments/examples/tlsf-init-test/main.c.
It seems to solve the problem in both embedded (only tried the samr21) and native. ATM it's a rough sketch.

Interestingly, newlib's `preinit_array` is a cross file array!!

closed 2019-09-10 by stale[bot]

jcarrano, 2019-09-10:
Duplicate of #5796