RIOT-OS/RIOT #4491: pkg: tlsf: stats!

OlegHahm, 2015-12-16:
Similar to monitoring the stack usage with `ps` there should be some equivalent for heap usage when working with TLSF which reports the currently and maximum heap usage. (Particular the latter is of interest and should be rather simple to implement.)

Kijewski, 2015-12-16:
TLSF contains debug functions (which I removed in my patch). It should be easy enough to un-remove them and hook them up to shell calls.

OlegHahm, 2015-12-18:
Hm, which functions are you referring to?

P.S. Side question, did you understand which effect it would have to set `SL_INDEX_COUNT_LOG2` to a smaller number?

Kijewski, 2015-12-18:
> Hm, which functions are you referring to?

https://github.com/OlegHahm/tlsf/commit/5aa4b7759790c71e23d69aec867cc6cd35bd13fd#diff-b93798e7787e2d8eab313a92609c49e0L694

With this iteration you get the usable and used memory sans overhead.

> P.S. Side question, did you understand which effect it would have to set SL_INDEX_COUNT_LOG2 to a smaller number?

I think I knew what it meant at one point, but forgot since then. :) I would have to read into it again, but I don't have the time right now.

closed 2016-05-27 by OlegHahm