RIOT-OS/RIOT #4494: core: mutex_t::val could memorize holding thread id, not a boolean

Kijewski, 2015-12-16:
For debugging purposes it would be nice if `mutex_t::val` stored the current thread id, not a boolean. The test if the mutex is held would remain the same: `mutex_t::val == 0` → `mutex_t::val == KERNEL_PID_UNDEF` (`KERNEL_PID_UNDEF == 0`), so I see no extra overhead. The locking would be no more difficult either: `atomic_set_to_one(&mutex.val)`(which uses atomic_cas)  → `atomic_cas(&mutex.val, 0, sched_active_pid)`.

OlegHahm, 2015-12-18:
Makes sense to me. I think to remember that it was like that in older versions.

kaspar030, 2015-12-18:
It was. I don't remember why we changed that.

ISR's would need special handling when locking a mutex.

Kijewski, 2015-12-18:
Tracked it down to this "ancient" pre-RIOT diff: https://github.com/RIOT-OS/RIOT/commit/8b242c74a800a1281466179ec8e682b13fff1178#diff-bade665089346285998d082f3857199aL49

Kijewski, 2016-04-26:
Closed. Obsolete.

closed 2016-04-26 by Kijewski